ARMmbed/uvisor #46: Inconsistent coding style

jserv, 2015-09-16:
Comparing to other ARM mbed projects such as [mbed-drivers](https://github.com/ARMmbed/mbed-drivers) and [mbed-client](https://github.com/ARMmbed/mbed-client), I found that the existing coding style in `uvisor` is a bit inconsistent. For example, the keyword `if` in C program statement usually lacks of corresponding space in `uvisor`, but both `mbed-drivers` and `mbed-client` have the explicit space: `if (...)`.

I just wonder if there is the documentation about coding style globally applied to all mbed projects.

hugovincent, 2015-09-16:
Hi @jserv -- good spot, you're right. We have some inconsistencies across mbed OS wrt code style, including here. We have [this document](https://developer.mbed.org/teams/SDK-Development/wiki/mbed-sdk-coding-style) about the coding style in mbed classic, but it needs some updates for mbed OS.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-908

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1041

meriac, 2016-06-24:
(closed, all new code has been added with new convention, pre-existing code changed as we proceed)

closed 2016-06-24 by meriac