ARMmbed/uvisor #63: Allow configuration of 3 parameters to assemble $CPU

AlessandroA, 2015-10-07:
Hardware-specific code for uVisor lives in dedicated folders. Currently these are simply named after the 2 platforms that we support at the moment.

To make porting easier, we should change the `Makefile` and `Makefile.rules` formats so that the symbol `$CPU` is assembled from at least 3 parameters:
```make
CPU := $(ARCH)$(FAMILY)$(CHIP)
```

Targets that leave one or more of those fields empty get less granularity, but everything still works.

---

In general we can expect a porting to look like this:
- a new folder named `lowercase($(ARCH))` contains hw-specific code;
- the hw-specific code contains `#ifdef`s depending on `$(FAMILY)` and `$(CPU)`; we expect those to be very few, since uVisor only relies on few hw-specific parameters (memories boundaries, usually);
- the `release` rule is available for more complex release scripts; for example, in a given hw-specific `$(ARCH)` folder a `release-all` rule could iterate over all `$(ARCH)`s and over all `$(CPU)`s to create release binaries for all possible targets in the family.

hugovincent, 2015-10-07:
(FWIW, `CPU` might be a confusing name... perhaps MCU or partnumber or SKU or something is better?)

AlessandroA, 2015-10-08:
@hugovincent yup, MCU sounds good to me

meriac, 2016-02-17:
Fixed by https://github.com/ARMmbed/uvisor/pull/142

closed 2016-02-17 by meriac