ARMmbed/uvisor #65: STM32F4 ISR default handler is not UVISOR_NAKED and UVISOR_NORETURN

akselsm, 2015-10-09:
Without `UVISOR_NAKED` the compiler (especially in debug mode) introduces push/pop operations; the pop operation is cut off by the `bx lr` instruction in `unvic_isr_mux()` and the `r4-r11` registers might be broken as a consequence.

```asm
00001c24 <isr_default_handler>:
    1c24:       b480            push    {r7}
    1c26:       af00            add     r7, sp, #0
    1c28:       df10            svc     16
    1c2a:       df10            svc     16
    1c2c:       4770            bx      lr                 <---- returns before popping
    1c2e:       46bd            mov     sp, r7
    1c30:       f85d 7b04       ldr.w   r7, [sp], #4
    1c34:       4770            bx      lr
    1c36:       bf00            nop
```

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-907

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1040

closed 2015-10-20 by AlessandroA