ARMmbed/uvisor #70: Move debug funcionality to debug box

meriac, 2015-10-21:
- uVisor is supposed to have as little as possible special knowledge about the target and it peripherals
- uVisor must still support complex debugging like over meshed wireless networks, IPv6 Ethernet with SSL or USB CDC serial.

Solution:
- Debug communication and debug LED handling will be moved into a uVisor box
- Whenever uVisor will print messages, it will issue a de-privileged call to the dedicated box that handles the output.
- whenever the remote system wants to communicate to uVisor, the debug box will call into uVisor to deliver just the payload.
- as a result uVisor does not need to know about complex protocols, but is still able to interact with remote systems in a debug session.
- the debug box might be able to debug other boxes if these boxes allow to do so (similar to Android where an app needs to explicitly allow debugging in its manifest)
- we stop uVisors attack surface from growing by reducing privileged code
- uVisors security properties are very comparable across a multitude of system: security verifications become transferable across systems

See also https://github.com/ARMmbed/uvisor/pull/64

adbridge, 2016-02-25:
Will this debug box be available in both debug and release code? As well as outputting whatever debug information is required, what actions are taken wrt code execution? Will execution be stopped dead at that point, or will there be some form of flag that is passed over the API to determine the action?

meriac, 2016-02-25:
@adbridge Yes, the debug box will always be available, just the verbosity will be much lower in release mode

JaredCJR, 2016-03-01:
As I know,USARTx_Handler runs in privileged mode,but uvisor `vIRQ Handler` runs in unprivileged mode.I am not really sure about  how the mechanism works.
Therefore,I would like to know where is the entry when `vIRQ` happened in order to know where uvisor should perform saving the context for debug_box.I will try to implement it.

daniel-cesarini-tridonic-com, 2018-10-15:
@meriac @adbridge any news regarding this?
We would be interested in having a backtrace (stacktrace) printed in mbed_die. Do you think it will available anytime soon?

Thanks.

adbridge, 2018-10-18:
@daniel-cesarini-tridonic-com Unfortunately uvisor is no longer supported as of mbed-os 5.10 so there will obviously be no further updates on it. The security functionality going forward will be provided by PSA.