ARMmbed/uvisor #87: Context switch bug

meriac, 2015-10-28:
- Fix the interrupt bug related to tailchained IRQ that happen during a context switch.
- Affects STM32 and EFM32GG - probably also MK64F

Acceptance criteria:
- create test to reproduce the behaviour
- verify the bug if fixed in the context of the test

rainierwolfcastle, 2015-10-29:
ARM Internal Ref: IOTSFW-1124

AlessandroA, 2015-11-27:
This bug has been addressed with https://github.com/ARMmbed/uvisor/issues/113.

The problem was not related to interrupts tailchaining as it was thought at the beginning. Instead, a particular interrupt configuration allowed an IRQ to be enabled (`vIRQ_EnableIRQ`) even before being registered (`vIRQ_SetVector`).

Although this is a pattern that is not recommended for NVIC interrupts, some missing checks in uVisor made it hard to detect and debug. With https://github.com/ARMmbed/uvisor/issues/113 those checks are now in place, and an error is issued when NVIC flags for an IRQ are changed before registering the corresponding ISR.

@akselsm As for the EFM32GG, the problem was a missing interrupt registration in the GPIO HAL initialization. As described in the `uvisor-lib` [documentation](https://github.com/ARMmbed/uvisor-lib#interrupt-management), interrupts need to be registered before being enabled, as without registration uVisor cannot ensure ownership and protection from other secure boxes.
```C
vIRQ_SetVector(SOME_IRQn, (uint32_t) &SOME_IRQHandler);
vIRQ_EnableIRQ(SOME_IRQn);
```

closed 2015-11-27 by AlessandroA