ARMmbed/uvisor #125: [K64F] A secure box with an empty ACL list does not get the background ACLs

AlessandroA, 2016-01-14:
In `vmpu_load_boxes()` we have a `for` loop that scans ACLs and calls the function `vmpu_acl_add()` for each of them.

In K64F this function further calls `vmpu_aips_add()`, which
* adds the relevant AIPS slot and
* **also adds ACLs from box 0 to the ACL list of the currently parsed box**

This last point creates the bug: If the ACL list for a secure box is empty, then the `vmpu_acl_add()` will never be called and hence (for the K64F) the box 0 ACLs for AIPS slots are never applied to other boxes.

For reference:
* [Call to `vmpu_acl_add()`](https://github.com/ARMmbed/uvisor/blob/master/core/system/src/mpu/vmpu.c#L169)
* [Box 0 ACLs are added to other boxes in `vmpu_aips_add()`](https://github.com/ARMmbed/uvisor/blob/master/core/system/src/mpu/vmpu_freescale_k64_aips.c#L104)

---

Possible **solution**:
* Create a new function that gets called after ACLs parsing: `vmpu_add_common_acls()`
    * Implemented differently for ARM and Freescale MPUs
    * All shared box 0 ACLs can be accounted for after private ACLs have been parsed
    *  *sharing box 0 ACLs* is decoupled from *having own ACLs*

@meriac 
@Patater

avri-schneider, 2016-01-18:
See my comment here: https://github.com/ARMmbed/uvisor/pull/127#discussion-diff-49997361

AlessandroA, 2016-01-18:
Fixed in https://github.com/ARMmbed/uvisor/pull/128

closed 2016-01-18 by AlessandroA