ARMmbed/uvisor #133: Instructions in The uVisor as a yotta Module section does not work

SecurityOfThings2016, 2016-01-27:
Instructions in The uVisor as a yotta Module section does not work. 
yotta link command gives following error:
error: No module.json file.
error: The current directory does not contain a valid module.

SecurityOfThings2016, 2016-01-28:
The issue is observed both in Ubuntu and MacOS.

SecurityOfThings2016, 2016-01-28:
I managed  to find the workaround for this problem. I did git clone of https://github.com/ARMmbed/uvisor.git into uvisor folder and git clone of https://github.com/ARMmbed/uvisor-lib.git into uvisor/release folder. with this setup, I managed to run instructions in porting guide to build my version of uvisor-lib. It appears that there is permission issues with repositories to use with git clone --recursive https://github.com/ARMmbed/uvisor.git

meriac, 2016-01-28:
uvisor is not a yotta module. Please refer to the uvisor documentation (Debug/Porting) - the mbed interface to uvisor is uvisor-lib (dedicated repo with a precompiled uvisor version).

Uvisor-lib is included as the release folder/submodule in the uvisor repository for convenience. uvisor itself is OS independent.

The other issue mentioned in previous comment is a duplicate of https://github.com/ARMmbed/uvisor/issues/134

closed 2016-01-28 by meriac