ARMmbed/uvisor #134: Issue with porting guide

SecurityOfThings2016, 2016-01-28:
upon execution of command git clone --recursive git@github.com:ARMMbed/uvisor.git throws following error:
Cloning into 'uvisor'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

SecurityOfThings2016, 2016-01-28:
I managed  to find the workaround for this problem. I did git clone of https://github.com/ARMmbed/uvisor.git into uvisor folder and git clone of https://github.com/ARMmbed/uvisor-lib.git into uvisor/release folder. with this setup, I managed to run instructions in porting guide to build my version of uvisor-lib. It appears that there is permission issues with repositories to use with git clone --recursive https://github.com/ARMmbed/uvisor.git

meriac, 2016-01-28:
Please ensure to have your hosts ssh key registered on github and --recursive cloning will work as expected.

closed 2016-01-28 by meriac