ARMmbed/uvisor #154: Failed to build uvisor

JaredCJR, 2016-02-18:
Failed to build uvisor on the [latest commit(master branch)](https://github.com/ARMmbed/uvisor/commit/cef53f88dc5b4ec97474404d754f9d901f47ccc0).
At `uvisor/platform/stm32`
Commands:`$make clean release`

Did I forget to update something?
```
rm -f stm32.map stm32.elf stm32.bin stm32.asm\
      stm32.linker gdb.script source.c.tags JLink.log\
        ../../release/source/stm32/version.txt
find . ../../core -iname '*.o' -exec rm -f \{\} \;
arm-none-eabi-gcc -mcpu=cortex-m4 -march=armv7e-m -mthumb -Os -DNDEBUG -g3 -Wall -Werror -DARCH_MPU_ARMv7M -DCONFIGURATION_ -DPROGRAM_VERSION=\"0.9.10-55-gcef5-dirty\" -I. -Iinc -I../../core -I../../core/cmsis/inc -I../../core/system/inc -I../../core/system/inc/mpu -I../../core/debug/inc -I../../core/lib/printf -ffunction-sections -fdata-sections -fno-exceptions  -c -o ../../core/system/src/benchmark.o ../../core/system/src/benchmark.c
In file included from inc/config.h:25:0,
                 from ../../core/uvisor.h:25,
                 from ../../core/system/src/benchmark.c:17:
inc/configurations.h:65:2: error: #error "Unrecognized uVisor configuration. Check your Makefile."
 #error "Unrecognized uVisor configuration. Check your Makefile."
  ^
In file included from ../../core/uvisor.h:25:0,
                 from ../../core/system/src/benchmark.c:17:
inc/config.h:31:2: error: #error "Currently only STM32 devices with a CCM are supported."
 #error "Currently only STM32 devices with a CCM are supported."
  ^
In file included from ../../core/uvisor.h:31:0,
                 from ../../core/system/src/benchmark.c:17:
../../core/cmsis/inc/core_generic.h:71:2: error: #error "Unsupported ARM core. Make sure CORE_* is defined in your workspace."
 #error "Unsupported ARM core. Make sure CORE_* is defined in your workspace."
  ^
In file included from ../../core/system/inc/svc_cx.h:22:0,
                 from ../../core/system/inc/svc.h:22,
                 from ../../core/system/inc/unvic.h:21,
                 from ../../core/system/inc/system.h:21,
                 from ../../core/uvisor.h:74,
                 from ../../core/system/src/benchmark.c:17:
../../core/system/inc/mpu/vmpu.h: In function 'vmpu_sram_addr':
../../core/system/inc/mpu/vmpu.h:37:33: error: 'SRAM_ORIGIN' undeclared (first use in this function)
     return (((uint32_t) addr >= SRAM_ORIGIN) &&
                                 ^
../../core/system/inc/mpu/vmpu.h:37:33: note: each undeclared identifier is reported only once for each function it appears in
../../core/system/src/benchmark.c: In function 'benchmark_configure':
../../core/system/src/benchmark.c:30:9: error: 'CoreDebug' undeclared (first use in this function)
         CoreDebug->DEMCR |= (1 >> 24);
         ^
../../core/system/src/benchmark.c:33:9: error: 'DWT' undeclared (first use in this function)
         DWT->CYCCNT = 0;
         ^
../../core/system/src/benchmark.c: In function 'benchmark_start':
../../core/system/src/benchmark.c:45:5: error: 'DWT' undeclared (first use in this function)
     DWT->CYCCNT = 0;
     ^
../../core/system/src/benchmark.c: In function 'benchmark_stop':
../../core/system/src/benchmark.c:50:12: error: 'DWT' undeclared (first use in this function)
     return DWT->CYCCNT - g_benchmark_overhead;
            ^
../../core/system/src/benchmark.c:51:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
<builtin>: recipe for target '../../core/system/src/benchmark.o' failed
make: *** [../../core/system/src/benchmark.o] Error 1
```

And the [tutorial to build uvisor seems to be old.](https://github.com/ARMmbed/uvisor#the-uvisor-as-a-yotta-module)
The `k64f` directory no longer exists.
```
# assuming you are developing in ~/code, select the correct platform in the code tree
cd ~/code/k64f/uvisor
```

I have seen the `Warning: Enabling Debug Features on uVisor`,and it says that the current uvisor does not have `Debug Features`.

AlessandroA, 2016-02-18:
Hi @JaredCJR we are in the process of refactoring uVisor, which means that the documentation is currently out of date. Apologies for that -- We are fixing it.

The refactored version of uVisor currently does not support *some* debug features. In particular, it does not use blinking LED patterns to assess a fault, and it does not map addresses to peripheral names (that is, it doesn't have knowledge of the detailed device [memory map](https://github.com/ARMmbed/uvisor/blob/uvisor_with_memory_map_debug/stm32f4/uvisor/debug/stm32f4_memory_map.c) any more).

If you want to work with the pre-refactoring version of uVisor please use the **temporary** branch https://github.com/ARMmbed/uvisor/tree/uvisor_with_memory_map_debug in which the documentation still applies and the debug features are still there. Please note that the branch will be discontinued and deleted as soon as the documentation is fixed on `master` and the debug features restored.

JaredCJR, 2016-02-18:
Thanks for your fast reply!

AlessandroA, 2016-02-18:
No problem! I'm closing this one, feel free to re-open it if you still encounter the issue.

closed 2016-02-18 by AlessandroA