ARMmbed/uvisor #157: Release mode binaries aren't actually built in release mode

akselsm, 2016-02-22:
Trying to run `make all`, I noticed that the binaries that get output to the "release" folder are the exact same size as the ones in the "debug" folder. Also, the "release" binaries seem to still have the semihosting code intact, which requires a debugger to be connected for programs to run.

It looks like `clean` is not run between building debug and release, causing make/gcc to skip recompiling the binaries, therefore linking the old "debug" versions instead. This means that `-DNDEBUG` is not applied to the compile step in release mode.

This is also a problem when building multiple configurations. Since the source files aren't recompiled, any defines set in `configurations.h` won't actually be applied to the given configuration, you'll just end up with multiple copies of whatever configuration was built first.

The following build string successfully builds a single release mode binary without semihosting code: `make BUILD_MODE=release PLATFORM=<my_platform> clean <MY_CONFIG>`

meriac, 2016-02-22:
@AlessandroA

AlessandroA, 2016-02-22:
Thanks for reporting @akselsm .
This bug was due to my mistake in not checking the binaries from a single run of make but only singularly after single-configuration runs.

commit 327a925d (2016-02-22): Fix bug in makefile for debug/release builds

If the objects file are not removed between single builds, they are
reused across builds for the same configuration (debug/release) and
across builds for different configurations in the same platform.

Removing them makes sure that a new build is always triggered in the 2
cases above.

Fixes: #157 ("Release mode binaries aren't actually built in release mode")
PR #158 (2016-02-22): Fix bug in makefile for debug/release builds
If the objects file are not removed between single builds, they are
reused across builds for the same configuration.

Removing them always trigger a new build, for each configuration and
both in debug and release mode.

Fixes: #157 ("Release mode binaries aren't actually built in release mode")

@meriac @Patater @akselsm
closed 2016-02-22 by meriac