ARMmbed/uvisor #164: uvisor_read() gets wrong value

JaredCJR, 2016-02-24:
`uvisor_read` gets wrong value in this example.
I printed it out,and got `00000000000000000000000000000000`(32-bits)
However,with ST-LINK(debugger),I got different value.
And I think the ST-LINK is right.
This value is used to calculate how many FPB comparators does STM32F429i support.
```C
#define FPB_CTRL 0xE0002000
uvisor_read(debug_box,FPB_CTRL);
```
```
(gdb) p/t *0xE0002000
$1 = 1001100001
```

AlessandroA, 2016-02-26:
Hi @JaredCJR , the reason why you are observing a returned value of `0x0` is because currently the `uvisor_read` API is [not implemented](https://github.com/ARMmbed/uvisor/blob/master/core/system/src/mpu/vmpu.c#L215).

The `uvisor_read` API is what we generally refer to with *register-level gateway*. While the API signature for it has been released (see [here](https://github.com/ARMmbed/uvisor-lib/blob/master/uvisor-lib/register_gateway.h#L90)), we are still working on its implementation (this is the reason why it is not documented yet).

For reference: #38 

In the meantime, you can use the `uvisor_read{32, 16, 8}` APIs (see [here](https://github.com/ARMmbed/uvisor-lib/blob/master/uvisor-lib/secure_access.h#L70)), which give the same functionality. Please note that the mechanism those APIs use (a managed bus fault) will be discontinued and dropped when the register-level gateway is ready.

closed 2016-02-26 by meriac

JaredCJR, 2016-02-29:
Hi @AlessandroA ,I tried `uvisor_read32`,but it cause bus fault.
For example,
```C
uint32_t val = uvisor_read32((uint32_t*)CoreDebug->DEMCR);
```

The message from `$ nc localhost 2333`
```
HALT_ERROR(core/system/src/mpu/vmpu_armv7m.c#268): Access to restricted resource denied
```
I am wondering whether there is anything that I misunderstand.
Thanks for your reply!