ARMmbed/uvisor #173: Memory access macros do not work when optimization is turned on

AlessandroA, 2016-02-26:
**Bug description**
On the NXP FRDM-K64F it has been observed that turning on compiler
optimizations interferes with the content of the `r0` register in memory access
macros.

Although `r0` is in the register list of the inline asm used for the
`uvisor_write32()` API, it contains garbage at the moment of use. The bug is
observed in the form of a hard fault due to an invalid address (`r0`) in a `str`
instruction.

**To reproduce**
Create a yotta application that uses the `frdm-k64f-gcc` target. Use the
following source code:
```C
#include "mbed-drivers/mbed.h"

PwmOut led(D5);

void app_start(int, char **) {
    led = 0.5f;
}
```
and compile in release mode with `yotta build -r`. Run the program and observe the hard fault.
The fault comes from the `pwmout_write` function, in which a series of optimizations leads
to the following `str` instruction:
```asm
str r1, [r0]
dsb
nop
nop
nop
```
where `r0 = 0x0`. The instructions above implement the `uvisor_write32()` API.

AlessandroA, 2016-02-26:
Notes:
* uVisor does not need to be enabled to observe this bug.
* The bug is currently only observed on the K64F as that platform's low level APIs boil down to calls to uVisor memory access macros.

mjs-arm, 2016-03-10:
Here is  a (almost) minimal reproducer of the issue.  This code was reduced directly from the failing code above by folding away the various layers of macro and inline function:

typedef long unsigned int uint32_t ;
void foo(float value)
{
  register uint32_t r0 __asm__("r0") = 0x40038000u;
  register uint32_t r1 __asm__("r1") = value;
    __asm__ volatile( "str"" %[r1], [%[r0]]\n"
                      :
                      : [r0] "r" (r0),
                        [r1] "r" (r1));
}

The issue can be provoked in gcc-4.9 using the above code, the load of the address into r0 is corrupted by the injection of library call to truncate the float to an integer during the setup of r1.

Relevant parts of the gcc manual are:
https://gcc.gnu.org/onlinedocs/gcc-5.3.0/gcc/Extended-Asm.html#Extended-Asm
https://gcc.gnu.org/onlinedocs/gcc-5.3.0/gcc/Local-Reg-Vars.html#Local-Reg-Vars

The local register var extension in gcc is 'fragile', note the following guidance from the documentation link above:

> Defining such a register variable does not reserve the register; it remains available for other uses in places where flow control determines the variable's value is not live.

and

> Stores into local register variables may be deleted when they appear to be dead according to dataflow analysis. References to local register variables may be deleted or moved or simplified.

and

> As with global register variables, it is recommended that you choose a register that is normally saved and restored by function calls on your machine, so that library routines will not clobber it.

Noting that in this case r0 and r1 do not conform to the last recommendation.

I have not studied what is going on in gcc in detail but I suspect that the second local reg var assignment generates an implicit function call (for float to int truncation) since r0 is part of the call convention, data flow analysis notes that the value in r0 is dead and (legitimately) deletes the assignment.

There are several ways of resolving this. AFAICT there is no reason that this code needs to hardwire the use of r0 and r1 therefore the most obvious solution would be to simply remove the local register vars and let gcc's register allocator handle appropriate selection of registers for the inline asm.  This approach will also give better code; studying a number of instance of this idiom in the generated code it is evident that the r0 r1 hardwire is injecting an unnecessary mov in most occurrences of the macro.

The workaround committed, so far as I can tell, is not a robust workaround, rather in perturbs the compiler sufficiently that the observed failure goes latent. It is quite possible (and likely) that other uses of this macro will result in broken code.

AlessandroA, 2016-03-10:
Thanks for looking into this @mjs-arm ! I will study this in more detail, but as a first comment note that we need the `UVISOR_ASM_MEMORY_ACCESS` macros to perform a "signature" read or write, that is, a `ldr` or `str` operation with a well defined and known opcode (including the operands).

The reason is that currently we use such reads and writes to trigger "controlled" bus faults, which are parsed by the uVisor to decide whether an access can be granted or not.

For reference:
* [Opcode signatures](https://github.com/ARMmbed/uvisor/blob/master/core/system/inc/mpu/vmpu.h#L48).
* [Access macros](https://github.com/ARMmbed/uvisor/blob/master/core/uvisor_exports.h#L133) (the `UVISOR_NOP_GROUP` is used to account for imprecise bus faults).

This is only limited to the K64F platform, due to a known limitation. You'll find more information on this issue [here](https://github.com/ARMmbed/mbed-hal-k64f/pull/6).

It is currently a requirement for us to be able to use those registers explicitly, although we could use different ways to accomplish that — for example, with an explicit assignment to `r0` and `r1` inside the volatile assembly:

```C
asm volatile(
    "mov r0, %[addr] \n"
    "mov r1, %[value] \n"
    "str r1, [r0] \n"
:: [addr] (r) (addr), [value] (r) (value)
);
```

although this would infer the `mov`s even when not needed (e.g. if `addr`/`value` are already in those registers).

Of course `r0` and `r1` can be replaced with other registers, but must be hardcoded for the reasons above. Any other suggestion for this implementation is welcome.

mjs-arm, 2016-03-11:
Looking at this further I notice the following text in the local register variables gcc manual link above:

"""Warning: In the above example, be aware that a register (for example r0) can be call-clobbered by subsequent code, including function calls and library calls for arithmetic operators on other variables (for example the initialization of p2). In this case, use temporary variables for expressions between the register assignments:"""

However, if the example we have here is modified to use the trick suggest in the manual, gcc still screws it up.  I'll raise an issue with the gcc folks, but a quick fix backported to 4.9 and 5.0 isn;t going to help.  

The only robust workaround I can see right now is the one  you propose in your previous comment, but note that in that example gcc needs to be told that the asm clobbers r0 and r1 (imagine what would happen if gcc had the value of addr in r1 and value in r0 on entry, or if gcc had other values live in r0 / r1 over the asm ;-)

AlessandroA, 2016-11-22:
Closing as not relevant any more (we moved from these APIs to the Register-Level Gateway APIs).

closed 2016-11-22 by AlessandroA