ARMmbed/uvisor #185: Allow secure callbacks to box 0

AlessandroA, 2016-03-04:
Currently secure gateways to box 0 are not allowed.

We should implement a *secure callback* that allows switching context to box 0. This callback would look like any other secure gateway but will allow runtime-determined function pointers to be given as a destination for the callback.

AlessandroA, 2016-03-04:
@meriac What about the use model? Secure gateways are usually not called explicitly. It's an API that under the hood issues a sgw to itself. In this case,

I guess in this case we want boxes to explicitly ask for a box 0 callback, right? As in "I want to call this function in the context of the main box, not mine." (use case: untrusted functions from other libraries).

meriac, 2016-06-08:
Will be replaced by remote procedure calls.

closed 2016-06-08 by meriac