ARMmbed/uvisor #207: ARMv7M MPU alignment bits check failure

ccli8, 2016-04-29:
Hi,
I am porting uvisor to our target board Nuvoton nu-mbed-NUC472, and meet this check failure on running the uvisor-helloworld sample. I try to connect uvisor debug mechanism with VCOM and get the following log:
```
__uvisor_config.flash_start=0x00000000
FLASH_ORIGIN=0x00000000
__uvisor_config.sram_start=0x20000000
SRAM_ORIGIN=0x20000000
uVisor mode: 2
uvisor_ram : @0x20000000 (8192 bytes) [config]
             (0x20000000 (8192 bytes) [linker]
MPU.REGIONS=8
MPU.ALIGNMENT=0x0000001F
MPU.ALIGNMENT_BITS=5
HALT_ERROR(core/system/src/mpu/vmpu_armv7m.c#652): ARMv7M_MPU_ALIGNMENT_BITS are inconsistent with actual MPU alignment
```
The log points out the check failure is located in vmpu_armv7m.c. If I skip the check failure, things can go well.
```
void vmpu_arch_init(void)
{
    ......
    /* Perform sanity checks. */
    if (ARMv7M_MPU_REGIONS != ((MPU->TYPE >> MPU_TYPE_DREGION_Pos) & 0xFF)) {
        HALT_ERROR(SANITY_CHECK_FAILED, "ARMv7M_MPU_REGIONS is inconsistent with actual region count.\n\r");
    }
    if (ARMv7M_MPU_ALIGNMENT_BITS == vmpu_bits(aligment_mask)) {
        HALT_ERROR(SANITY_CHECK_FAILED, "ARMv7M_MPU_ALIGNMENT_BITS are inconsistent with actual MPU alignment\n\r");
    }
```

AlessandroA, 2016-05-11:
Hi @ccli8, is this still an issue?

ccli8, 2016-05-11:
@AlessandroA , this is still an issue. Do you make any modification relevant to it?

meriac, 2016-05-12:
@ccli8 - please send me the content of the MPU->TYPE registers, from the looks of it you need to adjust ARMv7M_MPU_REGIONS to the actual amount of regions on your platform.

Can you please determine the output of "vmpu_bits(aligment_mask)" (printf, debugging etc.)

ccli8, 2016-05-13:
@meriac Content of MPU->TYPE (supported MPU data regions) and vmpu_bits(aligment_mask) have given as above. You can check with code fragment as below:

> uvisor/core/system/src/mpu/vmpu_armv7m.c

```
void vmpu_arch_init(void)
{
    uint32_t aligment_mask;

    /* Disable the MPU. */
    MPU->CTRL = 0;
    /* Check MPU region alignment using region number zero. */
    MPU->RNR = 0;
    MPU->RBAR = MPU_RBAR_ADDR_Msk;
    aligment_mask = ~MPU->RBAR;
    MPU->RBAR = 0;

    /* Initialize round-robin slot pointer. */
    g_mpu_slot = ARMv7M_MPU_RESERVED_REGIONS;

    /* show basic mpu settings */
    DPRINTF("MPU.REGIONS=%i\r\n", (uint8_t)(MPU->TYPE >> MPU_TYPE_DREGION_Pos));
    DPRINTF("MPU.ALIGNMENT=0x%08X\r\n", aligment_mask);
    DPRINTF("MPU.ALIGNMENT_BITS=%i\r\n", vmpu_bits(aligment_mask));
```
Log of DPRINTF above:
```
MPU.REGIONS=8
MPU.ALIGNMENT=0x0000001F
MPU.ALIGNMENT_BITS=5
```
In our mcu, ARMv7M_MPU_ALIGNMENT_BITS and vmpu_bits(aligment_mask) are both 5, and so check is failed:
```
if (ARMv7M_MPU_ALIGNMENT_BITS == vmpu_bits(aligment_mask)) {
    HALT_ERROR(SANITY_CHECK_FAILED, "ARMv7M_MPU_ALIGNMENT_BITS are inconsistent with actual MPU alignment\n\r");
}
```
```
HALT_ERROR(core/system/src/mpu/vmpu_armv7m.c#652): ARMv7M_MPU_ALIGNMENT_BITS are inconsistent with actual MPU alignment
```
Besides, why do you require ARMv7M_MPU_ALIGNMENT_BITS to be 5? As I know, in ARMv7M MPU, region size must be a power of 2 and base address of the region must be aligned to its region size. So why 32-aligned?

PR #222 (2016-05-25): Fix 2 ARMv7-M sanity check bugs
Two bugs fixed:

* Wrong sanity check for MPU alignment bits in ARMv7-M. Fixes #207.
* Wrong offset in flash/SRAM address check when the memory origin is different from `0x0`.

@meriac @niklas-arm @Patater
commit dd50cf8e (2016-05-25): Fix vMPU alignment check for ARMv7-M

Fixes #207 "ARMv7M MPU alignment bits check failure"
closed 2016-05-25 by meriac