ARMmbed/uvisor #235: RTOS is running with uVisor privileges

Patater, 2016-06-03:
When we integrated uVisor and an RTOS, we did not yet deprivilege the RTOS. All RTOS-specific code, like mutexes, the scheduler, timers, and so forth currently run with uVisor privileges. Some uVisor-to-RTOS glue code in uvisor-lib also runs with uVisor privileges. This is not final and we must fix it.

Ideally, no code would run privileged outside the reproducibly built uVisor binary. This enables us to make security promises because all privileged code would run inside a the reproducible uVisor binary.

Furthermore, removing uVisor privileges from the RTOS will reduce the attack surface of systems built with uVisor.

jdfarm, 2016-08-16:
This is particularly critical because much of the RTOS makes no effort at security whatsoever. For example, SVC_Handler trusts r12 as a function pointer to branch to while privileged. PoC below.

```
box_main:
    ldr r12, = privileged
    svc 0
[...]
privileged:
    // muhahahaha
    mov r0, 0
    ldr r1, = MPU_CTRL
    str r0, r1

```

jdfarm, 2016-08-16:
(Also, just for reference, note the massive security hole in the form of the unprotected osEventObs function pointers in rt_osEventObserver.c)