ARMmbed/uvisor #312: Following steps in QuickStart identically results in undefined reference to __wrap__calloc_r when compiling

sherrellbc, 2016-08-23:
I have been playing around with this one for too many hours. If one follows the steps outlined in the QuickStart markdown I am met with this every time. Is anyone else seeing this problem? 

**GCC Version**
arm-none-eabi-gxx :  (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

**Operating System**
Arch Linux; Kernel 4.7.0-1-ARCH 

`[user@host uvisor-example]$ mbed compile -m K64F -t GCC_ARM
Building project uvisor-example (K64F, GCC_ARM)
Scan: .
Scan: FEATURE_CLIENT
Scan: FEATURE_BLE
Scan: FEATURE_UVISOR
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_STORAGE
Scan: FEATURE_IPV6
Scan: FEATURE_IPV4
Scan: mbed
Scan: env
Link: uvisor-example
/home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: alignment 1 of symbol `__uvisor_ps' in ./mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(uvisor-output.o) is smaller than 4 in ./mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(disabled.o)
/home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/libc.a(lib_a-mprec.o): In function `_Balloc':
mprec.c:(.text._Balloc+0x22): undefined reference to `__wrap__calloc_r'
mprec.c:(.text._Balloc+0x3e): undefined reference to `__wrap__calloc_r'
collect2: error: ld returned 1 exit status
[ERROR] /home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: alignment 1 of symbol `__uvisor_ps' in ./mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(uvisor-output.o) is smaller than 4 in ./mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(disabled.o)
/home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/libc.a(lib_a-mprec.o): In function `_Balloc':
mprec.c:(.text._Balloc+0x22): undefined reference to `__wrap__calloc_r'
mprec.c:(.text._Balloc+0x3e): undefined reference to `__wrap__calloc_r'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/user/Desktop/code/uvisor-example/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./.build/K64F/GCC_ARM" in "/home/user/Desktop/code/uvisor-example"
`

AlessandroA, 2016-08-24:
Hi, thanks for reporting.

Could you please confirm the version of mbed OS you are using? You can provide the git commit SHA if you want (`cd mbed-os && git log --pretty=oneline --abbrev-commit | head -n 1`).

We have observed a regression in the way allocation functions are wrapped (hence the error you see), but that should have been introduced after the latest release.

The reason you see this is because at the moment the `mbed new` command, which is part of the tutorial, fetches the latest commit of ARMmbed/mbed-os, while it really should check the latest release (IMO). See https://github.com/ARMmbed/mbed-os/issues/2533 for reference.

The solution for the moment is to fetch a specific version of mbed OS:

```bash
$ cd ~/code/uvisor_example
$ cd mbed-os
$ mbed update mbed-os-5.1.2
```

Let me know if this helps. If the issue I linked is not solved, I'll instead update the quick-start guide to include this step.

sherrellbc, 2016-08-24:
Hello,

As for your first request, your statement was correct; `mbed new` seems to pull the latest revision. As of now that revision is: 

`22acfbf Merge pull request #2461 from svastm/serial_asynch_f3`

Your suggestion to manually check out at a stable release seems to work -- or at least the source is now compiling. This makes a great deal of sense considering that [This](https://github.com/ARMmbed/mbed-os-example-uvisor) example always compiled successfully for me. The mbed-os.lib file in that example points to the same revision that you suggested to be used above. 

However, although it seems to compile, it does not actually _work_. I can push the binary to the device (which is the FRDM K64F in my case) but nothing happens. This is the case if I follow the QuickStart guide or use the example code (linked above).

sherrellbc, 2016-08-24:
@AlessandroA , Thanks for looking into this and pointing out the obvious problem. However, the QuickStart guide still seems to be problematic and not properly working for the FRDM K64F development board. 

I followed your instructions regarding checking out a specific release, but the steps provided in the guide seem to be insufficient to produce a working binary. Perhaps there are implied steps I am missing? 

I know the guide did not mention steps like `mbed config root`. Perhaps I am missing more?

sherrellbc, 2016-08-25:
This example code is still broken. The [mbed-os-example-uvisor](https://github.com/ARMmbed/mbed-os-example-uvisor) code works if taken back to mbed-os-5.1.1, or commit [4740aa4d](https://github.com/ARMmbed/mbed-os-example-uvisor/commit/4740aa4d8762758192392b6a93dde15bd82367cf).

meriac, 2016-08-26:
Fixed in https://github.com/ARMmbed/mbed-os/pull/2534 and the latest [mbed-os-example-uvisor](https://github.com/ARMmbed/mbed-os-example-uvisor)

closed 2016-08-26 by meriac