ARMmbed/uvisor #315: uVisor seems to break when using NVIC_EnableIRQ or writing to MPU

sherrellbc, 2016-08-25:
I have been trying to determine the appropriate uVisor ACL such that I can use the Ethernet hardware on the FRDM K64F development board. After a few hours of debugging and tracing the problem I managed to determine two points of failure when using EthernetInterface.cpp and (a dependency thereof) the LWIP implementation; the root of the corresponding source tree can be found [here](https://github.com/ARMmbed/mbed-os/tree/master/features/net/FEATURE_IPV4/lwip-interface).

The first problem I found was that the LWIP implementation was trying to disable the MPU, which I thought might be problematic for uVisor.

`MPU->CESR &= ~MPU_CESR_VLD_MASK;`

https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip-eth/arch/TARGET_Freescale/hardware_init_MK64F12.c#L41

I have defined the MPU as an enabled memory range in my ACL, but that does not prevent the system from stopping on that instruction. 

`
const UvisorBoxAclItem g_main_acl[] = {     
         {SIM,    sizeof(*SIM),    UVISOR_TACLDEF_PERIPH}, 
         {OSC,    sizeof(*OSC),    UVISOR_TACLDEF_PERIPH}, 
         {MCG,    sizeof(*MCG),    UVISOR_TACLDEF_PERIPH}, 
         {PORTA,  sizeof(*PORTA),  UVISOR_TACLDEF_PERIPH}, 
         {PORTB,  sizeof(*PORTB),  UVISOR_TACLDEF_PERIPH}, 
         {PORTC,  sizeof(*PORTC),  UVISOR_TACLDEF_PERIPH}, 
         {PORTD,  sizeof(*PORTD),  UVISOR_TACLDEF_PERIPH}, 
         {PORTE,  sizeof(*PORTE),  UVISOR_TACLDEF_PERIPH}, 
         {RTC,    sizeof(*RTC),    UVISOR_TACLDEF_PERIPH}, 
         {LPTMR0, sizeof(*LPTMR0), UVISOR_TACLDEF_PERIPH}, 
         {PIT,    sizeof(*PIT),    UVISOR_TACLDEF_PERIPH}, 
         {SMC,    sizeof(*SMC),    UVISOR_TACLDEF_PERIPH}, 
         {UART0,  sizeof(*UART0),  UVISOR_TACLDEF_PERIPH}, 
         {I2C0,   sizeof(*I2C0),   UVISOR_TACLDEF_PERIPH},
         {SPI0,   sizeof(*SPI0),   UVISOR_TACLDEF_PERIPH},
 
         /* EthernetInterface */
         {ENET,   sizeof(*ENET),   UVISOR_TACLDEF_PERIPH}, 
         {EWM,    sizeof(*EWM),   UVISOR_TACLDEF_PERIPH}, 
};
`

---------

`
    /* Enables Ethernet interrupt and NVIC. */
    ENET_EnableInterrupts(base, config->interrupt);
    if (config->interrupt & (kENET_RxByteInterrupt | kENET_RxFrameInterrupt))
    {
        EnableIRQ(s_enetRxIrqId[instance]);
    }
    if (config->interrupt & (kENET_TxByteInterrupt | kENET_TxFrameInterrupt))
    {
        EnableIRQ(s_enetTxIrqId[instance]);
    }
    if (config->interrupt & (kENET_BabrInterrupt | kENET_BabtInterrupt | kENET_GraceStopInterrupt | kENET_MiiInterrupt |
                             kENET_EBusERInterrupt | kENET_LateCollisionInterrupt | kENET_RetryLimitInterrupt |
                             kENET_UnderrunInterrupt | kENET_PayloadRxInterrupt | kENET_WakeupInterrupt))
    {
        EnableIRQ(s_enetErrIrqId[instance]);
    }
`

https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c#L453

As it turns out, `EnableIRQ()` is an inline function defined in `fsl_common.h` as a link to an NVIC definition:

`
static inline void EnableIRQ(IRQn_Type interrupt)
{
#if defined(FSL_FEATURE_SOC_INTMUX_COUNT) && (FSL_FEATURE_SOC_INTMUX_COUNT > 0)
    if (interrupt < FSL_FEATURE_INTMUX_IRQ_START_INDEX)
#endif
    {
        NVIC_EnableIRQ(interrupt);
    }
}
`

https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_common.h#L182

I could not trace the NVIC definition down any further as there were a very large number of them. However, I did check out the generated source code in a disassembler and it seems that `NVIC_EnableIRQ` is being translated to an SVC 3 call:

-----

As a side note, I tried using `UVISOR_PERMISSIVE` for my main ACL, but the problem still persists.

sherrellbc, 2016-08-26:
I hooked SVC_Handler and confirmed that this function is not returning when called from the [NVIC_EnableIRQ](https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_common.h#L182) proxy.

sherrellbc, 2016-08-26:
I was think I was able to solve the problem related to `NVIC_EnableIRQ`, but I am still unable to read the MPU. 

A simple program that _only_ read from the MPU and has the MPU ACL does not work.

meriac, 2016-08-26:
@sherrellbc : Please don't access the MPU from unprivileged code. We need to figure why lwip insists on disabling the MPU - which is clearly a dirty hack that needs to be solved differently.

sherrellbc, 2016-08-26:
@meriac, I agree. Is there a way to read or write to this memory without a uVisor fault? I do not have access to a debugger at the moment, but when I try to even _read_ MPU (or FB) registers I get some sort of error and trap into the uVisor, or so I can only assume. Execute will halt on the read instruction. 

I observed that LWIP will break if the MPU is not disabled. That is, if the line is commented out _and_ uVisor is disabled LWIP will not work. It all works as expected, of course, if LWIP is allowed to disable the MPU.

meriac, 2016-08-26:
@sherrellbc writing to the MPU is out of question. We can't allow that for security reasons while uVisor is enabled. We need to find out why lwIP believes disabling the MPU is a good idea and fix the underlying problem without touching the MPU configuration from user mode.

sherrellbc, 2016-08-26:
@meriac, As you can see linked above I have opened an issue over at mbed since they own the LWIP implementation. As far as the other error I was seeing (related to `NVIC_EnableIRQ`), the LWIP implementation did not seem to register its interrupts with

sherrellbc, 2016-08-27:
Perhaps it is related to the necessary permissions for the ENET hardware

https://github.com/ARMmbed/uvisor/blob/master/core/system/inc/mpu/vmpu_freescale_k64_mem.h#L24

I submitted an issue over at mbed since they own the LWIP implementation.

geky, 2016-08-29:
It looks like on reset, the K64F MPU defaults to a single, full-access memory-region covering the entire address space.

That is, except for bus master 3, which is dedicated to the ENET hardware. Whoever implemented the original lwip port must have just disabled the MPU as a quick solution, since as far as I'm aware is has no impact on vanilla mbed.

Granting access to bus master 3 seems to work as an alternative to disabling the MPU completely:
``` cpp
MPU->RGDAAC[0] |= 0x007c0000;
```

If we don't want lwip to make system wide changes, disabling the MPU could be moved to the k64f's startup sequence. In this case, does disabling the MPU need to be conditionally defined on the presence of uVisor?

I'm unfamiliar with how uVisor handles DMA, but is there a method to grant access to bus master 3? This will be needed for the ethernet to operate.

k64f mpu datasheet:
https://developer.mbed.org/media/uploads/GregC/k64f_rm_rev2.pdf#d204e3a1310_d819e16

sherrellbc, 2016-08-30:
This is a fix that I have also been working on. I (temporarily) patched the uVisor code to give the above permission and it still does not work. The above permissions do, however, keep us from generating memory access exceptions and halting within uVisor. 

However, the ENET IRQs are not being triggered unless the MPU is entirely disabled, so the implementation is still a bit broken.

sherrellbc, 2016-08-31:
Confirmed with a managed switch that no Ethernet frames are going over the wire when uVisor is enabled.

meriac, 2016-08-31:
@sherrellbc :You need to make this change bfore uVisor is enabled in the boot process - see the [SystemInit](https://github.com/ARMmbed/mbed-os/blob/03d8ea8e96051c0fa2d0a081f9f84b9c86d9ffb2/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/system_MK64F12.c#L119) function.

meriac, 2016-08-31:
@sherrellbc : we have a patch for uVisor in the working. We'll release that by the end of this week (pending merge in mbed-os).

sherrellbc, 2016-08-31:
@meriac 

I tried making the `MPU->RGDAAC[0] |= 0x007c0000` change in [uVisor](https://github.com/ARMmbed/uvisor/blob/master/core/system/inc/mpu/vmpu_freescale_k64_mem.h#L34) and then manually deploying the generated archives to mbed-os. This change did not seem to work. 

`#define UVISOR_TACL_BACKGROUND 0x000827D0U | 0x007C0000`

Actually, since I already went this far to do a bit of testing I went ahead and gave wide-open privileges for DMA as well:

`#define UVISOR_TACL_BACKGROUND 0x000827D0U | 0x007C0000 | 0x0001F000`

None of that seemed to work, although this was admittedly a bruteforce attempt to just "see what happens".

meriac, 2016-09-02:
@sherrellbc : uVisor is for portability and practical reasons agnostic of the security impact of an ACL. uVisors concept is whitelisting - if you give something access, you must know best whether that's justified.

uVisor deals with conflicts between ACLs instead - let's say two applications wanting access to the same peripheral.

The way uVisor is designed, it's impossible to create an ACL during runtime - each application has to commit on the ACLs it requires passively at boot time by quoting them in the box configuration. 

This allows a future update server to deny signing firmware images that require access to critical components.

Another example why I believe this behaviour is good: Imagine the DMA engine only being accessible in one box - but exposing DMA-service via a secure API that verifies whether source/destination pointers are owned by the caller.

meriac, 2016-09-02:
@sherrellbc: I have an early version of my solution merged (https://github.com/ARMmbed/uvisor/pull/320). This allows ENET DMA transactions just to and from box-0 (default box) memories: Not affecting uVisor security any more. You still need to add an ACL for the ENET peripheral itself.

We are waiting for another pull request merged into mbed-os - but I will prepare an update example application for you so you have access earlier.

meriac, 2016-09-02:
@sherrellbc: The [mbed-os-example-uvisor](https://github.com/ARMmbed/mbed-os-example-uvisor/tree/dev) dev branch now points to my [mbed-os fork](https://github.com/meriac/mbed-os) with the required changes.

meriac, 2016-09-02:
@sherrellbc : The only thing that uVisor silently ignores is setting an unowned IRQ vector to NULL. If uVisor does not object, then the NVIC function did succeed.

meriac, 2016-09-02:
@sherrellbc : For the time being you are forced to use NVIC_SetVector for the ENET related IRQs. We are considering adding a fallback for legacy drivers to accommodate making these changes for drivers residing in box 0.

sherrellbc, 2016-09-02:
@meriac ,

The project does not compile once you include an `EthernetInterface` object and call `connect()` on it. It fails with:

```
Link: mbed-os-example-uvisor                                                                                                                                                                                                                 
/home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: alignment 1 of symbol `__uvisor_ps' in ../mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(uvisor-output.o) is smaller than 4 in ../mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(disabled.o)                                                                                                                                                                                                 
/home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region m_data_2 overflowed with stack and heap                                                                    
collect2: error: ld returned 1 exit status                                                                                                                                                                                                   
[ERROR] /home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: alignment 1 of symbol `__uvisor_ps' in ../mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(uvisor-output.o) is smaller than 4 in ../mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(disabled.o)                                                                                                                                                                                         
/home/user/.programs/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region m_data_2 overflowed with stack and heap                                                                    
collect2: error: ld returned 1 exit status     
```

I forked `mbed-os-example-uvisor` [here](https://github.com/sherrellbc/mbed-os-example-uvisor/tree/dev_ethernet) and included my changes.

sherrellbc, 2016-09-02:
@meriac 

The interrupts are now firing but RTX will error at run-time (when an ENET IRQ fires):
`RTX error code: 0x00000001, task ID: 0x00000000`

I forked the mbed-os-example-uvisor [here](https://github.com/sherrellbc/mbed-os-example-uvisor/tree/dev_ethernet). 

I removed the LED boxes; all of the memory allocation between each box was causing an error at link time. I also hooked the ENET interrupts to make sure they were getting triggered; although they _are_ getting triggered now, it is at this point that we exit with error.

meriac, 2016-09-06:
@sherrellbc box 0 has the default OS-provided stack - uVisor only deals with configuration of stacks for secure boxes. One note on your [example code](https://github.com/sherrellbc/mbed-os-example-uvisor/blob/dev_ethernet/source/main.cpp#L82-L83): From all I can see the NVIC_SetVector/NVIC_GetVector calls need to happen before instantiating the ethernet interface.

sherrellbc, 2016-09-06:
@meriac 

I tried increasing ISR_STACK_SIZE but, alas, it does not fix the problem. What could uVisor be doing that could cause an RTX stack overflow error? There are no such issues when uVisor is disabled.

sherrellbc, 2016-09-06:
@meriac 

Diving a bit further into the ENET Hal layer I found that it is relatively simple. When an ENET IRQ is triggered (which _is_ working) the call tree looks like (for Rx): 

[ENET_Transmit_IRQHandler](https://github.com/ARMmbed/mbed-os/blob/b396436432d055a5c4a7d0364a1a52177a61bfcf/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c#L1693)
↓
[ENET_Transmit_IRQ_Handler](https://github.com/ARMmbed/mbed-os/blob/b396436432d055a5c4a7d0364a1a52177a61bfcf/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c#L1610)
↓
[ethernet_callback](https://github.com/ARMmbed/mbed-os/blob/db99e726e006ca3260769d6cd444f51eefec1708/features/net/FEATURE_IPV4/lwip-interface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c#L138)
↓
[enet_mac_rx_isr](https://github.com/ARMmbed/mbed-os/blob/db99e726e006ca3260769d6cd444f51eefec1708/features/net/FEATURE_IPV4/lwip-interface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c#L128)

This final function calls [`sys_sem_signal()`](https://github.com/ARMmbed/mbed-os/blob/db99e726e006ca3260769d6cd444f51eefec1708/features/net/FEATURE_IPV4/lwip-interface/lwip-sys/arch/lwip_sys_arch.c#L268) on a semaphore that is being [`sys_arch_sem_wait`](https://github.com/ARMmbed/mbed-os/blob/db99e726e006ca3260769d6cd444f51eefec1708/features/net/FEATURE_IPV4/lwip-interface/lwip-sys/arch/lwip_sys_arch.c#L251)ed on by a thread at [packet_rx](https://github.com/ARMmbed/mbed-os/blob/db99e726e006ca3260769d6cd444f51eefec1708/features/net/FEATURE_IPV4/lwip-interface/lwip-eth/arch/TARGET_Freescale/k64f_emac.c#L387).

The stack overflow (RTX 0x01) is happening before the ISR can post to the semaphore at wake the thread.

sherrellbc, 2016-09-07:
I updated my [fork](https://github.com/sherrellbc/mbed-os-example-uvisor/tree/dev_ethernet) (of your fork/branch) with to print more run-time information without overflowing the stack. I also included a macro to try with and without uVisor enabled. There seems to be some problem with system mutexes when uVisor is enabled that I cannot track down. 

Your patch in [libraries/](https://github.com/meriac/mbed-os/commit/2cdb0e3f7c1a5426462e5f8e4de8ce71b2cea911) needs to be in [features/](https://github.com/meriac/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip-eth/arch/TARGET_Freescale/hardware_init_MK64F12.c#L41) instead.

sherrellbc, 2016-09-08:
Manually increasing [OS_MAINSTKSIZE](https://github.com/ARMmbed/mbed-os/blob/master/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c#L86) did not seem to help.

sherrellbc, 2016-09-13:
@meriac 

Getting ENET to work under uVisor is central to a project I am currently working on. Do you have any insight into what may be causing the mutex failure?

meriac, 2016-09-13:
@sherrellbc please contact me directly at the email address mentioned in my github user profile.

sherrellbc, 2016-09-15:
@meriac,
I have sent you an email. Thanks for the interest.

sherrellbc, 2016-09-19:
I do not mind continuing to investigate this issue; I have simply run out of places to look. Insight into where to investigate further is welcomed if anyone can provide suggestions.

meriac, 2016-09-19:
@sherrellbc One thing that we observed recently is that mallocs might file due to under-dimensioned heap sizes. These kinds of problems turn into weird followup-bugs that look like uVisor faults. To identify them I would put an assertion into malloc that breaks into the debugger whenever malloc would return a null pointer.

Also you must run the debug version of uVisor to get uVisor "blue screens" with detailed information on the cause. See our [debug documentation](https://github.com/ARMmbed/uvisor/blob/master/docs/api/DEBUGGING.md) for further information.

sherrellbc, 2016-09-21:
@meriac,

Thank you for the suggestion. Although not related to this issue in anyway, I was unable to get the debug session working correctly given the instructions on that page. Is it up to date? I have since posted an SO question [here](http://stackoverflow.com/questions/39617746/debugging-mbed-os-5-using-j-link-opensda-firmware-and-gdb). Please advise.

sherrellbc, 2016-09-21:
@meriac,

Although unrelated to this git issue entirely, I was not able to successfully debug uVisor using the documentation provided. I have since posted an SO question **[here]()** detailing my trail and error. Is the documentation up to date?

sherrellbc, 2016-09-22:
@meriac 

I was trying to leverage the semihosting connection and log information, but it seems this results in a fault as shown above. If I remove these statements then we are _actually_ getting a Bus Fault

> ***********************************************************
>                     BUS FAULT
> ***********************************************************
> 
> * CFSR : 0x00008200
> 
> * BFAR : 0x009B0394
> 
> * MPU FAULT:
>   Slave port:       0
>   Address:          0x000B0394
>   Faulting regions: 
>     R00: 0x00000000 0xFFFFFFFF 0x000007D0 0x00000001
>   Master port:      0
>   Error attribute:  Data WRITE (user mode)
> 
> * MEMORY MAP
>   Address:           0x009B0394
>   Region/Peripheral: [not available]
>     Base address:    0x009B0394
>     End address:     0x009B0394
> 
> * EXCEPTION STACK FRAME
>   Exception from unprivileged code
>     psp:     0x20004C60
>     lr:      0xFFFFFFFD
>   Exception stack frame:
>     psp[07]: 0x21000000 | xPSR
>     psp[06]: 0x0002071C | pc
>     psp[05]: 0x00020839 | lr
>     psp[04]: 0x00001D7C | r12
>     psp[03]: 0x009B0390 | r3
>     psp[02]: 0x00000001 | r2
>     psp[01]: 0x00000000 | r1
>     psp[00]: 0x009B0390 | r0
> 
> * MPU CONFIGURATION
>   CESR: 0x80815101
>        Slave 0    Slave 1    Slave 2    Slave 3    Slave 4
>   EAR: 0x000B0394 0x00000000 0x00000000 0x00000000 0x00000000
>   EDR: 0x80000003 0x00000000 0x00000000 0x00000000 0x00000000
>        Start      End        Perm.      Valid
>   R00: 0x00000000 0xFFFFFFFF 0x000007D0 0x00000001
>   R01: 0x00000000 0x000334DF 0x0010000D 0x00000001
>   R02: 0x20000000 0x2003001F 0x00180017 0x00000001
>   R03: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R04: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R05: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R06: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R07: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R08: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R09: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R10: 0x00000000 0x0000001F 0x00000000 0x00000000
>   R11: 0x00000000 0x0000001F 0x00000000 0x00000000
> 
> ***********************************************************
> 
> HALT_ERROR(./core/system/src/mpu/vmpu_freescale_k64.c#126): Access to restricted resource denied

sherrellbc, 2016-09-23:
@meriac ,

Apparently something else recently fixed has had the side-effect of allowing ENET to properly work, (#320)[https://github.com/ARMmbed/uvisor/pull/320] alone presented all of the above effects. Confirmed to work as of 0.25.0:

[2a42255bba42271366703ce083bcab37374bcce6](https://github.com/ARMmbed/mbed-os/commit/2a42255bba42271366703ce083bcab37374bcce6)

Iceberg1988, 2016-11-25:
Hi @sherrellbc and @meriac,

I have nearly the same problems as described above https://github.com/ARMmbed/uvisor/issues/315#issuecomment-243429214
I can compile and run code which contains a TCPSocket and the EthernetInterface without active uVisor. I get the connection and can communicate with the TCP socket. But with active uVisor I get the following error message at runtime:
`HALT_ERROR(./core/system/src/unvic.c#178): IRQ 84 is unregistered; state cannot be changed`

Here is the exact point, where uVisor crashs: [fls_enet.c#L476](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c#L476)
mbed-os is at master: https://github.com/ARMmbed/mbed-os/commit/aeabcc947294f2a6cbe1e808c030779c7c460f10 with uVisor Version v0.26.1

After I registered the interrupts by adding
`NVIC_SetVector(s_enetRxIrqId[instance], ENET_Receive_IRQHandler);`
in [fsl_enet.c#L476](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c#L476) and
`NVIC_SetVector(s_enetTxIrqId[instance], ENET_Transmit_IRQHandler);`
[fsl_enet.c#L480](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c#L480) I added the prototypes by inserting 
`void ENET_Transmit_IRQHandler(void);`
`void ENET_Receive_IRQHandler(void);`
in [fsl_enet.c#L222](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c#L222)
Then uVisor throws no error anymore, but the ethernet interface don't get a IP-Address like described above.

On 23.Sep this Issue here is closed after it properly worked with https://github.com/ARMmbed/uvisor/pull/320 ? I already tryed it with this version, but get the same problems.
Can somebody give me a Link to a working project with ethernet and uVisor working? Maybe there is some problem with my code? Or is this still a issue of uVisor and mbed?

sherrellbc, 2017-01-19:
@Iceberg1988 

I can confirm no functional issues on the K64F platform with the [latest](https://github.com/ARMmbed/mbed-os/commit/34c1facf42a174f47fdf9002cd8c6bf10ac41744) release. Of course, a workaround for the above was developed and is still in effect; you must declare the appropriate ACLs in the public, or main, box. Attempting to isolate all ENET access to a box other than 0 remains broken. You can still _use_ the ENET hardware from other boxes, you just cannot attempt to reserve it as a private resource. The associated HALT error is as follows:

```
***********************************************************
                    HARD FAULT
***********************************************************

* FAULT SYNDROME REGISTERS

  HFSR: 0x40000000
  --> FORCED: another priority escalated to hard fault.

* EXCEPTION STACK FRAME
  Exception from privileged code
    msp:     0x1FFF1EF8
    lr:      0xFFFFFFF1
  Exception stack frame:
    msp[07]: 0x0100000F | xPSR
    msp[06]: 0x00001176 | pc
    msp[05]: 0x00026507 | lr
    msp[04]: 0xFFFFFFFF | r12
    msp[03]: 0x00001175 | r3
    msp[02]: 0x00000033 | r2
    msp[01]: 0x0001ED9D | r1
    msp[00]: 0x00000033 | r0

* MPU CONFIGURATION
  CESR: 0x00815101
       Slave 0    Slave 1    Slave 2    Slave 3    Slave 4
  EAR: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
  EDR: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
       Start      End        Perm.      Valid
  R00: 0x00000000 0xFFFFFFFF 0x000007D0 0x00000001
  R01: 0x00000000 0x0003307F 0x0010000D 0x00000001
  R02: 0x20000000 0x2003001F 0x00180017 0x00000001
  R03: 0x1FFF2080 0x1FFF249F 0x0000001E 0x00000000
  R04: 0x1FFF2500 0x1FFF4A9F 0x0000001E 0x00000000
  R05: 0x00000000 0x0000001F 0x00000000 0x00000000
  R06: 0x00000000 0x0000001F 0x00000000 0x00000000
  R07: 0x00000000 0x0000001F 0x00000000 0x00000000
  R08: 0x00000000 0x0000001F 0x00000000 0x00000000
  R09: 0x00000000 0x0000001F 0x00000000 0x00000000
  R10: 0x00000000 0x0000001F 0x00000000 0x00000000
  R11: 0x00000000 0x0000001F 0x00000000 0x00000000

***********************************************************

HALT_ERROR(./core/vmpu/src/kinetis/vmpu_kinetis.c#155): Cannot recover from a hard fault.
```

The minimum ACL set to support ENET seems to be:

```
static const UvisorBoxAclItem g_main_box_acls[] = { 
    /* System requirements for timing, monitor calls, etc */
    {SIM,   sizeof(*SIM),   UVISOR_TACLDEF_PERIPH},
    {PIT,   sizeof(*PIT), UVISOR_TACLDEF_PERIPH},    
    {OSC,   sizeof(*OSC),   UVISOR_TACLDEF_PERIPH},   
    {MCG,   sizeof(*MCG),   UVISOR_TACLDEF_PERIPH},

    /* Ethernet Hardware */
    {ENET,  sizeof(*ENET),  UVISOR_TACLDEF_PERIPH},
    {PORTA, sizeof(*PORTA), UVISOR_TACLDEF_PERIPH}, 
    {PORTB, sizeof(*PORTB), UVISOR_TACLDEF_PERIPH},
    {PORTC, sizeof(*PORTC), UVISOR_TACLDEF_PERIPH},
};

```

Of course, you must also manually register the ENET interrupts as shown above.

sherrellbc, 2017-02-09:
@meriac 

Has there been any further progress to fix ENET only working from box 0 mentioned at https://github.com/ARMmbed/uvisor/issues/315#issuecomment-244363591 above?