ARMmbed/uvisor #317: The RPC middleware function should pass the in-flash gateway address as argument

AlessandroA, 2016-08-26:
Currently the RPC gateway implementation passes the destination function when calling `rpc_fncall_(a)sync(r0, r1, r2, r3, fp)`, but this makes the gateway verification much more difficult.

In fact, it is currently necessary to scan all detectable RPC gateways and check which one of them has the destination function.

Instead, we will pass the address of the in-flash gateway and use that inside uVisor to obtain the destination box and destination function for the RPC.

That is the same pointer that ends up in the enqueued message.

This also implies that having the gateway pointers in a discoverable section is not needed any more.

The change would keep backwards compatibility with the idea that `rpc_fncall_(a)sync` might be used without uVisor. In that case the `fp` argument is the actual destination function.

sherrellbc, 2016-08-30:
Where are these rpc_fncall_* functions defined? I searched all branches and only found references to them in the `rpc_gateway.h` header file.

meriac, 2016-09-18:
@AlessandroA @Patater - I assume this can be closed?

Patater, 2016-09-19:
This issue was created based on an old version of the PR, so nothing ever needed changed from what was in the current PR.

What this issue requests is not yet merged into dev or master, though, so maybe keep it open until https://github.com/ARMmbed/uvisor/pull/308 closes if you want to track it. Otherwise, feel free to close.

Patater, 2016-09-27:
Fixed by #308

closed 2016-09-27 by AlessandroA