ARMmbed/uvisor #319: The v7-M MPU ports fail with an ACL-related fault after the MPU interface refactoring

AlessandroA, 2016-09-02:
Following #294 the v7-M MPU ports of uVisor fail on the latest mbed OS (5.1.2) with the following halt error:

```text
uVisor mode: 2
uvisor_ram : @0x20000140 (7872 bytes) [config]
             @0x20000140 (7872 bytes) [linker]
bss_boxes  : @0x20002000 (8320 bytes) [config]

box stack segment start=0x20002000 end=0x20002000 (length=8320)
MPU.REGIONS=8
MPU.ALIGNMENT=0x0000001F
MPU.ALIGNMENT_BITS=5
HALT_ERROR(./core/system/src/mpu/vmpu_armv7m_mpu.c#178): inferred ACL's (0x003F) don't match exptected ACL's (0x0037)
```

This is probably due to an ACL configuration inconsistency.

Reported by @niklas-arm (EFM32GG) and @fvincenzo (Beetle).

nchong, 2016-09-09:
I think this could be from `vmpu_arch_init_hw()` in `vmpu_armv7m.c`. The upper SRAM static acl is requested with `UVISOR_TACL_UEXECUTE` but no corresponding `UVISOR_TACL_SEXECUTE`. This is unrepresentable by a v7-M RASR. The acl must be both user/supervisor executable or both non-executable.

asensaoui, 2016-09-12:
I've tried this and it resolved the problem 
`vmpu_mpu_set_static_acl(
        1,
        (uint32_t) __uvisor_config.sram_start,
        total_size,
        UVISOR_TACLDEF_DATA | UVISOR_TACL_SEXECUTE | UVISOR_TACL_UEXECUTE,
        subregions_disable_mask
    );`
I don't know if it is the right thing to do.

niklas-arm, 2016-09-27:
This was fixed in https://github.com/ARMmbed/uvisor/pull/335.

closed 2016-09-27 by AlessandroA