ARMmbed/uvisor #325: v7-M MPU port issues with external SRAM

ccli8, 2016-09-14:
**My development platform**
Target: NUMAKER_PFM_NUC472
Toolchain: gcc 4.9.3
mbed-os-example-uvisor (c440d07129293b21cd76fd0359322ee9a050398c)
mbed-os (e9d0fbd178332a681d37568c4ae8d4737cb6419a)
uvisor (d949d43bf96cabc638fa876882be05978d238b80)

I am porting uvisor on the platform NUMAKER_PFM_NUC472, which has 64 KB internal SRAM (0x20000000) and 1 MB external SRAM (0x60000000). I test with the sample [mbed-os-example-uvisor](https://github.com/ARMmbed/mbed-os-example-uvisor) and meet the first problem as #319. If I skip it and go along, I meet the memory check error:
```
uVisor mode: 2
uvisor_ram : @0x20000000 (8192 bytes) [config]
             @0x20000000 (8192 bytes) [linker]
bss_boxes  : @0x20002000 (6272 bytes) [config]

box stack segment start=0x20002000 end=0x20003880 (length=6272)
MPU.REGIONS=8
MPU.ALIGNMENT=0x0000001F
MPU.ALIGNMENT_BITS=5
* MPU CONFIGURATION

  Background region enabled
  MPU bypassed @NMI, @HardFault
  MPU enabled

  Region Start      Size  XN AP  TEX S C B SRD      Valid
  0      0x00000000 128KB 0  110 000 0 0 0 00000000 1
  1      0x20000000 064KB 0  011 000 0 0 0 11111111 1
  2      0x00000000 000B  0  000 000 0 0 0 00000000 0
  3      0x00000000 000B  0  000 000 0 0 0 00000000 0
  4      0x00000000 000B  0  000 000 0 0 0 00000000 0
  5      0x00000000 000B  0  000 000 0 0 0 00000000 0
  6      0x00000000 000B  0  000 000 0 0 0 00000000 0
  7      0x00000000 000B  0  000 000 0 0 0 00000000 0

HALT_ERROR(./core/system/src/mpu/vmpu.c#234): Heap start pointer (0x60000020) is not in SRAM memory.
```
In my gcc linker script, due to just 64 KB internal SRAM (0x20000000), I locate heap and page heap in external SRAM (0x60000000), so I meet the check error. My problems are:

1. Would you have a plan to support external SRAM for uvisor on v7-M MPU?
2. I try to locate heap and page heap in internal SRAM but still fail to run [mbed-os-example-uvisor](https://github.com/ARMmbed/mbed-os-example-uvisor). Its memory footprint is above 64 KB SRAM. What is the minimum memory requirement to run uvisor on v7-M MPU?

meriac, 2016-09-18:
@ccli8 I created an example [memory footprint rendering](http://www.meriac.com/mbed/mbed-os-example-uvisor-irq/) for our [IRQ example](https://github.com/ARMmbed/mbed-os-example-uvisor-irq). You can see the static memory consumption of uVisor in SRAM and FLASH plus the page allocator reserved for the secure box (compile time configurable).

Regarding external S(D)RAM support: You need to understand though that a local attacker will have an easy game with that. For now you might consider using the external S(D)RAM for the default security box - and use internal SRAM from the secure boxes. For that you need to update the linker script and ensure that the uVIsor secure page allocator allocates default box memories from SDRAM and secure memories in the internal SRAM.
You can find an example close to your expected configuration in our [K64F port](https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/cmsis/TARGET_Freescale/TARGET_MCU_K64F/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld#L69-L70) where two memories are used, too.

closed 2016-09-18 by meriac

ccli8, 2016-09-23:
@meriac The [K64F port](https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/cmsis/TARGET_Freescale/TARGET_MCU_K64F/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld#L69-L70)'s two memory blocks are consecutive, and is not close to NUMAKER_PFM_NUC472, where its two SRAM blocks are inconsecutive:
```
K64F
0x1FFF0000-0x20000000
0x20000000-0x20030000
```
```
NUMAKER_PFM_NUC472
IRAM: 0x20000000-0x20010000 (64 KB)
XRAM: 0x60000000-0x60100000 (1 MB)
```
Now I locate box bss and page heap in IRAM as usual and heap in XRAM. With that layout, I meet the same check error as above:
```
uVisor mode: 2
uvisor_ram : @0x20000000 (8192 bytes) [config]
             @0x20000000 (8192 bytes) [linker]
bss_boxes  : @0x20002000 (8320 bytes) [config]

box stack segment start=0x20002000 end=0x20004080 (length=8320)
MPU.REGIONS=8
MPU.ALIGNMENT=0x0000001F
MPU.ALIGNMENT_BITS=5
* MPU CONFIGURATION

  Background region enabled
  MPU bypassed @NMI, @HardFault
  MPU enabled

  Region Start      Size  XN AP  TEX S C B SRD      Valid
  0      0x00000000 128KB 0  110 000 0 0 0 00000000 1
  1      0x20000000 064KB 0  011 000 0 0 0 00000111 1
  2      0x00000000 000B  0  000 000 0 0 0 00000000 0
  3      0x00000000 000B  0  000 000 0 0 0 00000000 0
  4      0x00000000 000B  0  000 000 0 0 0 00000000 0
  5      0x00000000 000B  0  000 000 0 0 0 00000000 0
  6      0x00000000 000B  0  000 000 0 0 0 00000000 0
  7      0x00000000 000B  0  000 000 0 0 0 00000000 0

HALT_ERROR(./core/system/src/mpu/vmpu.c#234): Heap start pointer (0x60000020) is not in SRAM memory.
```
If I remove the error check in `vmpu.c` and manually add XRAM into ACL list of main box in `mbed-os-example-uvisor`, I could run the sample. So what about remove the checks to support XRAM like our case? Or other suggestion?
```
   if (!__uvisor_config.heap_start || !vmpu_sram_addr((uint32_t) __uvisor_config.heap_start)) {
        HALT_ERROR(SANITY_CHECK_FAILED, "Heap start pointer (0x%08x) is not in SRAM memory.\n",
            (uint32_t) __uvisor_config.heap_start);
    }
    if (!__uvisor_config.heap_end || !vmpu_sram_addr((uint32_t) __uvisor_config.heap_end)) {
        HALT_ERROR(SANITY_CHECK_FAILED, "Heap end pointer (0x%08x) is not in SRAM memory.\n",
            (uint32_t) __uvisor_config.heap_end);
    }
```
```
#define MAIN_ACL(acl_list_name) \
    static const UvisorBoxAclItem acl_list_name[] = {     \
        {...}, \
        {(void *)0x60000000,    0x100000,    UVISOR_TACL_ACCESS}, \
    }
```