ARMmbed/uvisor #326: Interrupts cause memFault

asensaoui, 2016-09-15:
Sometimes, interrupts cause a Memmanag fault, not everytime, so i create a small program with only one interruption which is a button (sw1), here is the link to the source files : https://github.com/asensaoui/uvisor_interrupts 
( I ported my own board, it's not officially supported, and also I changed all the NVIC to __NVIC and made some modification to core_cm3.h (adding the CMSIS_NVIC_VIRTUAL ...) I've also changed this to bypass the bug within the initialization : 
`    /* Unlock the upper SRAM subregion only. */
    vmpu_mpu_set_static_acl(
        1,
        (uint32_t) __uvisor_config.sram_start,
        total_size,
        UVISOR_TACLDEF_DATA | UVISOR_TACL_EXECUTE,
        subregions_disable_mask
    );`
)

Here is the output of the debugger : 
396,635 @"***********************************************************\n"
396,648 @"                    MEMMANAGE FAULT\n"
396,680 @"***********************************************************\n"
396,693 @"\n"
396,726 @"* CFSR  : 0x00000001\n"
396,739 @"\r\n"
396,752 @"\r* MMFAR : 0xE000ED34\n"
396,766 @"\r\n"
396,779 @"\r* No MPU violation found\n"
396,792 @"\r\n"
396,805 @"\r* MEMORY MAP\n"
396,818 @"\r  Address:           0xE000ED34\n"
396,850 @"\r  Region/Peripheral: [not available]\n"
396,864 @"\r    Base address:    0xE000ED34\n"
396,896 @"\r    End address:     0xE000ED34\n"
396,909 @"\r    Before bitband:  [invalid]\n"
396,922 @"\r\n"
396,954 @"\r* EXCEPTION STACK FRAME\n"
396,967 @"  Exception from unprivileged code\n"
397,000 @"    psp:     0x2000A2A8\n"
397,033 @"\r    lr:      0xFFFFFFFD\n"
397,046 @"\r  Exception stack frame:\n"
397,059 @"    psp[08]: 0x2000A6BC | align\n"
397,072 @"    psp[07]: 0x00000000 | xPSR\n"
397,104 @"    psp[06]: 0x00000000 | pc\n"
397,117 @"    psp[05]: 0x1000BFCD | lr\n"
397,149 @"    psp[04]: 0x10001E9C | r12\n"
397,162 @"    psp[03]: 0x10009327 | r3\n"
397,175 @"    psp[02]: 0x00000000 | r2\n"
397,207 @"    psp[01]: 0x2000AA94 | r1\n"
397,220 @"    psp[00]: 0x00000400 | r0\n"
397,233 @"\n"
397,246 @"* MPU CONFIGURATION\n"
397,259 @"\r\n"
397,272 @"\r  Background region enabled\n"
397,285 @"\r  MPU bypassed @NMI, @HardFault\n"
397,298 @"\r  MPU enabled\n"
397,311 @"\r\n"
397,324 @"\r  Region Start      Size  XN AP  TEX S C B SRD      Valid\n"
397,356 @"\r  0      0x10000000 128KB 0  110 000 0 0 0 00000000 1\n"
397,369 @"\r  1      0x20000000 128KB 0  011 000 0 0 0 00000011 1\n"
397,401 @"\r  2      0x42100000 004KB 1  011 000 0 0 0 00000000 1\n"
397,414 @"\r  3      0x20002000 008KB 1  011 000 0 0 0 00100000 1\n"
397,427 @"\r  4      0x40020000 064B  1  011 000 0 0 0 00000000 1\n"
397,459 @"\r  5      0x40008000 008KB 1  011 000 0 0 0 00000000 1\n"
397,473 @"\r  6      0x20020000 128KB 1  011 000 0 0 0 00000000 1\n"
397,486 @"\r  7      0x4002C000 004KB 1  011 000 0 0 0 00000000 1\n"
397,518 @"\r\n"
397,531 @"\r***********************************************************\n"
397,563 @"\n"
397,576 @"HALT_ERROR(./core/system/src/mpu/vmpu_armv7m.c#157): Access to restricted resource denied\\
n"

I tried interrupts with with only one thread, and it works fine, so i thought maybe it's because the switch context.
To track the problem, I tried to print in and out to context_switch_in( ... ).
I observed that every time I push the button when the box owning the interrupt is active, I get the memFault, but when I press the button when another box is active, it works fine.

387,867 @"\rIRQ 24 registered to box 1\n"
387,880 @"\rIRQ 24 pending status cleared\n"
387,913 @"\rIRQ 24 enabled\n"
388,935 @"\rend context switch in to 0, from 1, now active box is 0\n"
388,969 @"end context switch in to 1, from 0, now active box is 1\n"
390,011 @"end context switch in to 0, from 1, now active box is 0\n"
390,046 @"end context switch in to 1, from 0, now active box is 1\n"
391,047 @"end context switch in to 0, from 1, now active box is 0\n"
391,121 @"end context switch in to 1, from 0, now active box is 1\n"
392,121 @"end context switch in to 0, from 1, now active box is 0\n"
392,176 @"end context switch in to 1, from 0, now active box is 1\n"
393,162 @"end context switch in to 0, from 1, now active box is 0\n"
393,235 @"end context switch in to 1, from 0, now active box is 1\n"
394,211 @"end context switch in to 0, from 1, now active box is 0\n"
394,304 @"end context switch in to 1, from 0, now active box is 1\n"
395,300 @"end context switch in to 0, from 1, now active box is 0\n"
395,394 @"end context switch in to 1, from 0, now active box is 1\n"
396,362 @"end context switch in to 0, from 1, now active box is 0\n"
396,475 @"end context switch in to 1, from 0, now active box is 1\n"
396,590 @"end context switch in to 1, from 1, now active box is 1\n"
396,622 @"\n"
396,635 @"***********************************************************\n"
396,648 @"                    MEMMANAGE FAULT\n"
396,680 @"***********************************************************\n"

I don't understand why we do a switch context when an interrupt occurs even if we are already in the box owning the interrupt?
I was thinking of disabling the interrupt whenever we are in the box owning that interrupt., but i don't know if it's really safe to do so, I tried something like that, it solved the problem partially, but the application wasn't working properly.

Thank you for your help.

CC @meriac @AlessandroA

asensaoui, 2016-09-15:
I did something so simple that solve the problem, actually, i deleted the Thread::wait within the box who owns the interrupt, and it works better, still don't understand why.

meriac, 2016-09-17:
@asensaoui I created a working [uVisor IRQ example](https://github.com/ARMmbed/mbed-os-example-uvisor-irq) - here are some observations on potential problems of the code you uploaded:
- You need to be aware that the global variables [you initialized](https://github.com/asensaoui/uvisor_interrupts/blob/master/source/box1.cpp#L35-L36) are in insecure memories - make sure to use the uVisor box context instead as in the new IRQ example.
- Probably you allow box 0 to run code with box 1 privileges as the streaming serial driver is used. Make sure to use RawSerial and allocate it inside the box instead. If you don't like malloc/new you can use the inplace version of the new operator instead.

I am closing this issue now as the IRQ example works as expected for me - if you still have trouble, please try to reproduce it on a fork of this example and tell me the platform you're using it on (it was unclear from your code example, so I had to assume K64F).

@AlessandroA

asensaoui, 2016-09-19:
Your code works fine for me, but in my program within the box where I defined the interrupt, I have a thread who does some stuff. I have a feeling like we can't interrupt a thread within a box with an interrupt owned by the same box, for example : 
We have a box who owns an interrupt X: 
while Main_box_thread() is doing his stuff, the X interrupt happens, after the end of the interrupt stuff, a mem fault is generated and the address of the pc is weird.
@meriac @AlessandroA

meriac, 2016-09-19:
@asensaoui can you please explain your setup in detail - what microcontroller and development board are you precisely using? Your problem description is too vague for me to identify your problem:

"while Main_box_thread() is doing his stuff, the X interrupt happens, after the end of the interrupt stuff, a mem fault is generated and the address of the pc is weird"

asensaoui, 2016-09-20:
I'm using a Max32550 cortex M3. 
This is a simple example https://github.com/asensaoui/uvisor_interrupts (updated my last code). 
My problem is when I push the button several times, it goes into a memmanag fault.
For now I have 2 suggestions who may cause the fault :

asensaoui, 2016-09-23:
I'm closing this because the PR #335 fixes the problem, nonetheless, IRQ nesting is broken, I can't get the right behavior when an interrupt calls another ones.

closed 2016-09-23 by asensaoui

niklas-arm, 2016-09-27:
Disabled IRQ nesting is tracked in #345.