ARMmbed/uvisor #357: PSP contains sometimes a wrong address

asensaoui, 2016-10-19:
When an Interrupt occurs, sometimes we get a wrong psp address, then it goes into a hardfault. 
Here is a snippet from the debugger showing the id of the active box the dst box and the value of the psp: 
 
875,787 @"*  box id: 1  dst box id : 1 \n"
875,820 @"*  lr: 0x100038EF sp: 0x20001F1C  psp : 0x20005048 ipsr: 0x0000000B \n"
875,852 @"IRQ 19 pending status cleared\n"
875,864 @"\r*  lr: 0x10003991 sp: 0x20001F4C  psp : 0x20007FD8 ipsr: 0x0000000B \n"
875,940 @"*  box id: 2  dst box id : 1 \n"
875,954 @"*  lr: 0x100038EF sp: 0x20001EDC  psp : 0x20008940 ipsr: 0x0000000B \n"
875,966 @"IRQ 19 pending status cleared\n"
875,979 @"\r*  lr: 0x10003991 sp: 0x20001F0C  psp : 0x20007FD8 ipsr: 0x0000000B \n"
876,117 @"*  box id: 2  dst box id : 1 \n"
876,130 @"*  lr: 0x100038EF sp: 0x20001F1C  psp : 0x20008940 ipsr: 0x0000000B \n"
876,163 @"IRQ 19 pending status cleared\n"
876,176 @"\r*  lr: 0x10003991 sp: 0x20001F4C  psp : 0x20007FD8 ipsr: 0x0000000B \n"
876,251 @"*  box id: 2  dst box id : 1 \n"
876,265 @"*  lr: 0x100038EF sp: 0x20001F1C  psp : 0x20008940 ipsr: 0x0000000B \n"
876,277 @"IRQ 19 pending status cleared\n"
876,310 @"\r*  lr: 0x10003991 sp: 0x20001F4C  psp : 0x20007FD8 ipsr: 0x0000000B \n"
876,366 @"*  box id: 2  dst box id : 1 \n"
876,380 @"*  lr: 0x100038EF sp: 0x20001F1C  psp : 0x20008940 ipsr: 0x0000000B \n"
876,392 @"IRQ 19 pending status cleared\n"
876,425 @"\r*  lr: 0x10003991 sp: 0x20001F4C  psp : 0x20007FD8 ipsr: 0x0000000B \n"
876,501 @"*  box id: 2  dst box id : 1 \n"
876,514 @"*  lr: 0x100038EF sp: 0x20001EC4  psp : 0x20005020 ipsr: 0x0000000B \n"
876,547 @"\n"
876,579 @"***********************************************************\n"
876,612 @"                    HARD FAULT\n"
876,624 @"***********************************************************\n"

Obviously, just before the hardfault, we must have in the psp something like 0x20008xxx, but we actually have an address within the stack of box 1 and this generates a Hardfault during the
 `src_sp = context_validate_exc_sf(__get_PSP())`.

I did finally succeed reproducing the problem but you have to be patient, because sometimes you need to push the button many many times. 
https://github.com/asensaoui/uvisor_example

Thanks,

@Patater

asensaoui, 2016-11-18:
I think the problem is due to : 
when we perform a thread switch, we don't set the PSP in the function switch_context, we set the PSP within HAL_CM3.S(rtos) after comparing os_tsk.run and os_tsk.new, the thing is if an ISR happens just after the thread switch and before we into the PendSV_Handler, we can get : 
active_box = the dst box (the one we switch into), but the PSP points to the stack of the previous box ()the one we just switched from) thats why the check line [unvic.c #L477](https://github.com/ARMmbed/uvisor/blob/master/core/system/src/unvic.c#L477) fails.

@Patater

asensaoui, 2016-11-22:
To fix this I did the following : 
I deleted the atomic call wrapper in Thread.c, and used the non atomic function and then, I disable irq before the `bl  rt_systick` and enable irq just before the `BX      LR                      /* Return to Thread Mode */ `   in HAL_CM3.S (rtos)
so I have the following code : 

HAL_CM3.S : 
```
/*-------------------------- PendSV_Handler ---------------------------------*/

#       void PendSV_Handler (void);

        .thumb_func
        .type   PendSV_Handler, %function
        .global PendSV_Handler
        .global Sys_Switch
PendSV_Handler:
        .ifdef  IFX_XMC4XXX
        .global PendSV_Handler_Veneer
PendSV_Handler_Veneer:
        .endif
        .fnstart
        .cantunwind

        BL      rt_pop_req

Sys_Switch:

        LDR     R3,=os_tsk
        LDM     R3,{R1,R2}              /* os_tsk.run, os_tsk.new */
        CMP     R1,R2
        BEQ     Sys_Exit

        

        MRS     R12,PSP                 /* Read PSP */
        STMDB   R12!,{R4-R11}           /* Save Old context */
        STR     R12,[R1,#TCB_TSTACK]    /* Update os_tsk.run->tsk_stack */

        PUSH    {R2,R3}
        BL      rt_stk_check            /* Check for Stack overflow */
        POP     {R2,R3}

        STR     R2,[R3]                 /* os_tsk.run = os_tsk.new */

        LDR     R12,[R2,#TCB_TSTACK]    /* os_tsk.new->tsk_stack */
        LDMIA   R12!,{R4-R11}           /* Restore New Context */
        MSR     PSP,R12                 /* Write PSP */

Sys_Exit:
        MVN     LR,#~0xFFFFFFFD         /* set EXC_RETURN value */
        .ifdef  IFX_XMC4XXX
        PUSH    {LR}
        POP     {PC}
        .else
+        ldr     r3, [r7, #12]
+        and.w   r3, r3, #1
+        cmp     r3, #0
+        cpsie   i
        BX      LR                      /* Return to Thread Mode */
        .endif

        .fnend
        .size   PendSV_Handler, .-PendSV_Handler

/*-------------------------- SysTick_Handler --------------------------------*/

#       void SysTick_Handler (void);

        .thumb_func
        .type   SysTick_Handler, %function
        .global SysTick_Handler
SysTick_Handler:
        .ifdef  IFX_XMC4XXX
        .global SysTick_Handler_Veneer
SysTick_Handler_Veneer:
        .endif
        .fnstart
        .cantunwind

+        cpsid   i
+        ldr     r0, [r7, #4]
        BL      rt_systick
        B       Sys_Switch
        

        .fnend
        .size   SysTick_Handler, .-SysTick_Handler
```
The Idea is to prevent ISRs when we are within the SysTick_handler.
I'm just wondering if there is anything wrong with this changes? 

@Patater @niklas-arm @meriac @AlessandroA

Patater, 2017-06-06:
@asensaoui

We have addressed this issue in uVisor v0.28.1, shipped with mbed OS 5.5. Please feel free to reopen this issue if it is still a problem for you.

closed 2017-06-06 by Patater