ARMmbed/uvisor #387: v7M box bss allocation is under-estimated

ccli8, 2016-12-20:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**uvisor sha:**
07dde17d610cca51f24135de9c65b15938f89f97

I am adding uvisor support for Nuvoton's new target board NuMaker-PFM-M487. By test with the example `mbed-os-example-uvisor`, I meet the error with uvisor debug message below:
```
uVisor mode: 2
uvisor_ram : @0x20000000 (8192 bytes) [config]
             @0x20000000 (8192 bytes) [linker]
bss_boxes  : @0x20002000 (16512 bytes) [config]

box stack segment start=0x20002000 end=0x20006080 (length=16512)
MPU.REGIONS=8
MPU.ALIGNMENT=0x0000001F
MPU.ALIGNMENT_BITS=5
* MPU CONFIGURATION

  Background region enabled
  MPU bypassed @NMI, @HardFault
  MPU enabled

  Region Start      Size  XN AP  TEX S C B SRD      Valid
  0      0x00000000 128KB 0  110 000 0 0 0 00000000 1
  1      0x20000000 128KB 0  011 000 0 0 0 00011111 1
  2      0x00000000 000B  0  000 000 0 0 0 00000000 0
  3      0x00000000 000B  0  000 000 0 0 0 00000000 0
  4      0x00000000 000B  0  000 000 0 0 0 00000000 0
  5      0x00000000 000B  0  000 000 0 0 0 00000000 0
  6      0x00000000 000B  0  000 000 0 0 0 00000000 0
  7      0x00000000 000B  0  000 000 0 0 0 00000000 0

box[0] ACL list:
ctx=1380 stack=0
box[1] ACL list:
        box[1] stack=1024 bss=9584 rounded=16384
HALT_ERROR(./core/vmpu/src/armv7m/vmpu_armv7m.c#352): memory overflow - increase uvisor memory allocation
```

I check with uvisor core and linker script, and find allocation for box bss is under-estimated. v7M MPU region base address requires to be a multiple of its size, and so uvisor has the following code fragment to guarantee per-box bss is aligned with its size.
**uvisor/core/vmpu/src/armv7m/vmpu_armv7m.c > vmpu_acl_stack:**
```
/* Check for correct context address alignment. Alignment needs to be a
     * muiltiple of the size. */
    if ((box_mem_pos & (size - 1)) != 0) {
        box_mem_pos = (box_mem_pos & ~(size - 1)) + size;
    }
```
But in linker script, the constraint of per-box bss base address alignment is not taken into consideration and so finally box bss is under-estimated.
**[uVisor Porting Guide for mbed OS](https://github.com/ARMmbed/uvisor/blob/master/docs/core/PORTING.md) > Lnker script:**
```
/* Protected uVisor boxes' static memories */
        . = ALIGN(32);
        __uvisor_bss_boxes_start = .;
        KEEP(*(.keep.uvisor.bss.boxes))
        . = ALIGN(32);
        __uvisor_bss_boxes_end = .;
```

meriac, 2017-01-03:
@ccli8 - a quick workaround is to declare an variable sized array in .keep.uvisor.bss.boxes to reserve the missing memory. Independent of that we will introduce a slightly better optimised runtime-ordering of boxes and box-rounding - but that won't change the underlying problem - that could only be fixed by adding a pre-linking step to do that automatically. At present we're not planning to do that.
To make the manual adding of memory easier we'll add a pre-processor macro for reserving memory if needed and we'll print the actually amount missing during boot. These changes will be added during the next 4-6 weeks.

@AlessandroA @niklas-arm @Patater

Patater, 2017-01-05:
This is related to https://github.com/ARMmbed/uvisor/issues/353

AlessandroA, 2017-03-08:
This is now fixed by [uVisor v0.27.0](https://github.com/ARMmbed/uvisor/releases/tag/v0.27.0). In particular:

* At runtime, uVisor will throw a developer-friendly error message if additional memory should be allocated for boxes.
* At runtime, boxes enumeration permutes box order so that memory consumption is minimized.

The feature will be available with the upcoming mbed OS 5.4 release. I hope this solves your issue, @ccli8.

closed 2017-03-08 by AlessandroA