ARMmbed/uvisor #392: A quick change in debugging documentation

asensaoui, 2017-01-06:
Box config called twice and we don't set up the heap_size : 
[debugging.md](https://github.com/ARMmbed/uvisor/blob/07dde17d610cca51f24135de9c65b15938f89f97/docs/api/DEBUGGING.md#the-debug-box)
```
/* Configure the debug box. */
UVISOR_BOX_NAMESPACE(NULL);
UVISOR_BOX_CONFIG(box_debug, UVISOR_BOX_STACK_SIZE); // box config called twice, and the heap size is not set up
UVISOR_BOX_MAIN(box_debug_main, osPriorityNormal, UVISOR_BOX_STACK_SIZE);
UVISOR_BOX_CONFIG(box_debug, acl, UVISOR_BOX_STACK_SIZE, box_context);
```

I have also some questions about the debug_box: 
Does the system reboot only in release mode and not in the debug one? If so, can I reboot it after halting the system in debug mode? 

Could we change the behavior of the system after catching an exception, like for example restarting just the box leading to the conflict?

Best regards,

adbridge, 2017-01-10:
ARM Internal Ref: IOTSEC-247

AlessandroA, 2017-01-19:
Hi @asensaoui thanks for reporting. #400 should fix the bug in the docs. As for your questions:

> Does the system reboot only in release mode and not in the debug one? If so, can I reboot it after halting the system in debug mode?

Since uVisor 0.26.0 the debug box does not reboot automatically anymore. Upon return from the `halt_error` handler, the system halts.

You can still reboot the system if you want, by calling the `NVIC_SystemReset()` API. This is only possible from the debug box. It works both in debug and release mode:

```C
static void halt_error(int reason) {
    printf("We halted with reason %i\r\n", reason);

    /* If we don't do anything, the system will halt upon return. */
    /* A debug box handler like this one can also decide to reboot the whole
     * system. This is only allowed from the debug box. */
    NVIC_SystemReset();
}
```

> Could we change the behavior of the system after catching an exception, like for example restarting just the box leading to the conflict?

We do not support the reboot of a single box yet, but it's definitely something we would like to look into in the future.

AlessandroA, 2017-01-19:
Fixed in #400.

closed 2017-01-19 by AlessandroA