ARMmbed/uvisor #423: mbedTLS fails with uVisor enabled due to memory allocation errors

sherrellbc, 2017-04-17:
I've created a [minimal example ](https://github.com/sherrellbc/mbedtls-uvisor-minimal/blob/master/main_box.cpp) that seems to work only when uVisor is not enabled. The failure occurs in the [mbedtls_pk_parse_key](https://github.com/sherrellbc/mbedtls-uvisor-minimal/blob/master/main_box.cpp#L183) function when parsing the private key. 

When using the "small" key (512 bits) I observe the following run-time error:
`mbedtls_pk_parse_key() failed: -0x3d10 (-15632): PK - Invalid key tag or value : BIGNUM - Memory allocation failed`

When using the "large" key (2048 bits) I observe a similar error:
`mbedtls_pk_parse_key() failed: -0x4210 (-16912): RSA - Key failed to pass the library's validity check : BIGNUM - Memory allocation failed`

For reference, the example code is compiled and executed on FRDM board. I have tried adjusting the [heap and stack](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld#L64) sizes, but it does not seem to help. We are currently working with mbed-os 5.3.4 and uVisor version 0.27.0.

Is there a known issue with running mbedTLS and uVisor concurrently?

AlessandroA, 2017-04-20:
Possibly related to https://github.com/ARMmbed/mbed-os/issues/3610. @sherrellbc we are currently investigating the issue.

clemej1, 2017-04-20:
I've been working with @sherrellbc on this, I can verify that it is probably related, but the workaround is not sufficient.  

Defining `MBED_HEAP_STATS_ENABLED` does allow us to get further on our internal code, but still fails with strange errors in `mbedtls_pk_parse_key()`, around a call to `mbedtls_mpi_grow( X, i + j )` at `bignum.c:1174`   ... which I'm guessing is still a potential memory allocation error based solely on the function name ending in `_grow()`. 

We've been testing with our full code, not our example code posted above.  We'll re-test with the example code and post results when we can, but it may take until tomorrow.

sherrellbc, 2017-06-25:
@AlessandroA Have there been any updates that resolve this problem?

Patater, 2017-06-26:
@sherrellbc We've fixed a bug in the mbed OS 5.5 release of uVisor where the box main thread stack and box heap were placed incorrectly, leading to heap corruption under certain conditions. It's possible that in your example, you really do run out of memory, but it could also be heap corruption, so it'd be worth retesting with mbed OS 5.5.

Thanks

sherrellbc, 2017-06-26:
@Patater The latest 5.5 release candidate seemed to solve the problem. Actually, the default heap size for the NXP FRDM board was too small (0x6000). Increasing this (actually _doubling_ it) prevented run-time allocation errors from occurring. Otherwise `mbedtls_calloc` will fail with `mbedtls_ssl_setup() failed: -0x7f00 (-32512): SSL - Memory allocation failed`.

closed 2017-07-17 by sherrellbc