ARMmbed/uvisor #435: Fail to compile uvisor for K64F board using GCC_ARM: flexible array member 'uvisor_pool::management_array' not at end of 'struct uvisor_rpc_t'

cregnec, 2017-06-14:
`mbed import mbed-os-example-uvisor`

[mbed] Importing program "mbed-os-example-uvisor" from "https://github.com/ARMmbed/mbed-os-example-uvisor" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #fc1836545dcc

`mbed compile -m K64F -t GCC_ARM --profile mbed-os/tools/profiles/debug.json -c`
Building project mbed-os-example-uvisor (K64F, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_STORAGE
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_NANOSTACK
Scan: FEATURE_LOWPAN_ROUTER
Scan: mbed
Scan: env
Compile [  0.3%]: Ethernet.cpp
[Error] pool_queue_exports.h@93,48: flexible array member 'uvisor_pool::management_array' not at end of 'struct uvisor_rpc_t'
[ERROR] In file included from ./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/api.h:25:0,
                 from ./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/interrupts.h:22,
                 from ./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/cmsis_nvic_virtual.h:20,
                 from ./mbed-os/cmsis/core_cm4.h:1517,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/MK64F12.h:286,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/fsl_device_registers.h:46,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/cmsis.h:10,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/objects.h:19,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/device.h:37,
                 from ./mbed-os/platform/platform.h:29,
                 from ./mbed-os/drivers/Ethernet.h:19,
                 from ./mbed-os/drivers/Ethernet.cpp:16:
./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/pool_queue_exports.h:93:48: error: flexible array member 'uvisor_pool::management_array' not at end of 'struct uvisor_rpc_t'
     uvisor_pool_queue_entry_t management_array[];
                                                ^
In file included from ./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/box_config.h:22:0,
                 from ./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/uvisor-lib.h:29,
                 from ./mbed-os/features/FEATURE_UVISOR/includes/uvisor-lib/uvisor-lib.h:42,
                 from ./mbed-os/cmsis/core_cmSecureAccess.h:45,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_clock.h:34,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_common.h:132,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_dspi.h:33,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_dspi_edma.h:33,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/objects.h:24,
                 from ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/device.h:37,
                 from ./mbed-os/platform/platform.h:29,
                 from ./mbed-os/drivers/Ethernet.h:19,
                 from ./mbed-os/drivers/Ethernet.cpp:16:
./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/rpc_exports.h:116:48: note: next member 'uvisor_pool_queue_entry_t uvisor_rpc_outgoing_message_queue_t::entries [8]' declared here
         uvisor_pool_queue_entry_t entries[slots]; \
                                                ^
./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/rpc_exports.h:137:9: note: in expansion of macro 'UVISOR_RPC_OUTGOING_MESSAGE_TYPE'
 typedef UVISOR_RPC_OUTGOING_MESSAGE_TYPE(UVISOR_RPC_OUTGOING_MESSAGE_SLOTS) uvisor_rpc_outgoing_message_queue_t;
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./mbed-os/features/FEATURE_UVISOR/includes/uvisor/api/inc/rpc_exports.h:141:16: note: in the definition of 'struct uvisor_rpc_t'
 typedef struct uvisor_rpc_t {
                ^~~~~~~~~~~~

[mbed] ERROR: "python" returned error code 1.

cregnec, 2017-07-20:
Actually, this was caused by  ARM GCC v7.1. Downgrade to v6 solved the problem.

closed 2017-07-20 by cregnec

thomasfangyi, 2017-08-01:
Is it possible to set standard to gnu11/gnu99 explicitly?
Static initialisation of flexible array member is a GNU extension, it should be able to compile without error.

See https://gcc.gnu.org/onlinedocs/gcc/Zero-Length.html#Zero-Length

Taiki-San, 2018-02-16:
Setting the standard explicitly to to a more recent version of C/C++ didn't fix the issue.