ARMmbed/uvisor #440: missing file "process_malloc.h" during compilation

gruenwaldi, 2017-06-20:
I created a new configuration for the NRF52832 controller for uVisor, by following the instructions from "https://github.com/ARMmbed/uvisor/blob/master/docs/core/PORTING.md" . Afterwards, without changing the actual code, the compiler gives me the following error:
***************************************************************
Compile [ 16.6%]: box_init.c
[Fatal Error] uvisor-lib.h@43,43: uvisor-lib/rtx/process_malloc.h: No such file or directory
[ERROR] In file included from ./mbed-os/features/FEATURE_UVISOR/source/rtx/box_init.c:17:0:
./mbed-os/features/FEATURE_UVISOR/includes/uvisor-lib/uvisor-lib.h:43:43: fatal error: uvisor-lib/rtx/process_malloc.h: No such file or directory
 #include "uvisor-lib/rtx/process_malloc.h"
                                           ^
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
****************************************************************

So I looked into mbed-os, but couldn't locate this file anywhere.
Even if I comment it out, in the "uvisor-lib.h" file, other files appear, which I cannot find. 
How can I proceed?

Patater, 2017-08-02:
Hi @gruenwaldi 

When adding new target support for uVisor, the uVisor import script located in mbed OS also needs to be updated. This is so that uVisor files will be copied to a location picked up by the mbed OS build system for your target. I suspect the build system isn't finding the header files because they weren't copied for your target.

Add your target to the target translation line in `features/FEATURE_UVISOR/importer/Makefile`. Refer to your target in `targets/targets.json` to see what to use here.
```diff
  # Translate between uVisor namespace and mbed namespace
- TARGET_TRANSLATION:=MCU_K64F.kinetis EFM32.efm32 STM32F4.stm32 ARM_BEETLE_SOC.beetle
+ TARGET_TRANSLATION:=MCU_K64F.kinetis EFM32.efm32 STM32F4.stm32 ARM_BEETLE_SOC.beetle NXP_NEW_TARGET.nxp_new_target
```

Patater, 2017-08-14:
@gruenwaldi Please reopen if this solution doesn't work for you.

Thanks

closed 2017-08-14 by Patater