ARMmbed/uvisor #500: Would interrupt be handled in realtime in uVisor ARMv8m ?

p03189, 2017-08-31:
In uvisor/core/system/src/core_armv8m/virq.c: virq_switch()
Only the interrupts which belongs to active box can be enabled. Because boxes are scheduled by uVisor ARM8vm in Round-Robin, an Interrupt may not be handled in realtime if it doesn't belong to active box. Is it true ?

Patater, 2017-08-31:
That's true. The current implementation protects bother boxes against a malicious box causing an interrupt storm by guaranteeing each box a slice of time in which to run.

p03189, 2017-08-31:
Got it ! Thank you.

In uvisor/core/system/src/box_init.c: box_init()
I also found that uVisor takes MSP_NS for boxes and RTX5 takes PSP_NS for threads. When a box is scheduled in and invoke any system call, for instance osDelay(), SVC handler will try to fetch invalid value from PSP. Will you add any patch for RTX5 ? Why would uVisor take MSP_NS for boxes ?

Patater, 2017-08-31:
Only the public box, box 0, has a valid PSP (as a result of RTX running in the public box). All other boxes run privileged (from their first entry, and could run unprivileged) and without RTX. RTX is not supported from secure boxes on v8-M.

Do you have some use cases in mind where you need or want RTX in secure boxes?

closed 2017-08-31 by Patater

p03189, 2017-09-01:
Hi @Patater ,

I don't have use case for this, we are in early stage to integrate uVisor-ARMv8m to our project.

Do you have any document or wiki page for uVisor-ARMv8m ? I only found some document for uVisor-ARMv7m.

Would systick_ns be disabled when a secure box (not public one) are activated to prevent disturbance from RTX's round robin ?

Thanks for your reply

Patater, 2017-09-01:
SYSTICK_NS state is saved and restored on box switch, so that other boxes can also use the peripheral.

p03189, 2017-09-01:
Thank you so much !