ARMmbed/uvisor #521: uvisor with mbedTLS

gylalahuo, 2017-10-16:
mbed Version：5.6.3
Board： K64F  
environment：ARM-GCC
when I use mbedTLS to realize https communicate with my Server，everything is OK ，but when I enable the uvisor，there is a error。
The problem is   mbedtls_ssl_setup() failed: -0x7f00 (-32512): SSL - Memory allocation failed，In mbedtls_ssl_setup()，there are double calloc（1，17771） funciton。
 so i try UVISOR_SET_PAGE_HEAP（）to increase the memory，it doesn't work。 then  i  try revise  the  MK64FN1M0xxx12.ld  ，and set __heap_size__ = 0xB000; this problem solve。but I think this method is incorrect。 Because at first when I don‘t  enable uvisor，everything is OK，so the heap_size is enough. If there some changes when I use uvsior，I think should use UVISOR_SET_PAGE_HEAP（） to solve problem rather than  revise  the  MK64FN1M0xxx12.ld

alzix, 2017-10-18:
Thank you for the input. We will look in to it.

Can you please share some additional info:
1. can you share with us your sample app the reproduces the problem?
2. Alternatively - please share uVisor configuration macros you have used per each box.

gylalahuo, 2017-10-19:
Hi alzix,
    I only enable a public box in main.cpp 
   the configuratiuon is:

#define MAIN_ACL(acl_list_name) \
    static const UvisorBoxAclItem acl_list_name[] = {     \
        {SIM,    sizeof(*SIM),    UVISOR_TACLDEF_PERIPH}, \
        {OSC,    sizeof(*OSC),    UVISOR_TACLDEF_PERIPH}, \
        {MCG,    sizeof(*MCG),    UVISOR_TACLDEF_PERIPH}, \
        {PORTA,  sizeof(*PORTA),  UVISOR_TACLDEF_PERIPH}, \
        {PORTB,  sizeof(*PORTB),  UVISOR_TACLDEF_PERIPH}, \
        {PORTC,  sizeof(*PORTC),  UVISOR_TACLDEF_PERIPH}, \
        {PORTD,  sizeof(*PORTD),  UVISOR_TACLDEF_PERIPH}, \
        {PORTE,  sizeof(*PORTE),  UVISOR_TACLDEF_PERIPH}, \
        {RTC,    sizeof(*RTC),    UVISOR_TACLDEF_PERIPH}, \
        {LPTMR0, sizeof(*LPTMR0), UVISOR_TACLDEF_PERIPH}, \
        {PIT,    sizeof(*PIT),    UVISOR_TACLDEF_PERIPH}, \
        {SMC,    sizeof(*SMC),    UVISOR_TACLDEF_PERIPH}, \
        {UART0,  sizeof(*UART0),  UVISOR_TACLDEF_PERIPH}, \
        {I2C0,   sizeof(*I2C0),   UVISOR_TACLDEF_PERIPH}, \
        {SPI0,   sizeof(*SPI0),   UVISOR_TACLDEF_PERIPH}, \
        {RNG,   sizeof(*RNG),   UVISOR_TACLDEF_PERIPH}, \
    }

   MAIN_ACL(g_main_acl);
   UVISOR_SET_MODE_ACL(UVISOR_ENABLED, g_main_acl);
   UVISOR_SET_PAGE_HEAP(8 * 1024, 4);

Thanks& Best Regards

orenc17A, 2017-10-19:
@gylalahuo we will need more information like:
1. compilation command
2. your main.c

if you could share a link to github repo that would be super

gylalahuo, 2017-10-22:
Hi
my main.cpp

 is in attach files
Look forward to your reply!
Thanks !

gylalahuo, 2017-10-22:
compilation  command is  "mbed compile -t GCC_ARM - m K64F"

gylalahuo, 2017-10-22:
the project url is https://github.com/gylalahuo/uvisor_mbedTLS.git

gylalahuo, 2017-10-23:
the problem is solved

gylalahuo, 2017-10-23:
still have the problem

alzix, 2017-10-23:
@gylalahuo ,
I suspect this issue is related to #453 ,
Please refer to [Debugging uVisor on mbed OS](https://github.com/ARMmbed/uvisor/blob/master/docs/lib/DEBUGGING.md) for enabling uVisor semihosting prints. Make sure whether the issue you are experiencing is RTC related (faulting address within 0x4003D000 range)

gylalahuo, 2017-10-27:
@alzix
hi alzix， 
Thanks for you answer. The problem about time（） is solved ,but i still don’t the reason about "SSL - Memory allocation failed",why  i revise the heap memory size in MK64FN1M0xxx12.ld can solve this problem，rather than use UVISOR_SET_PAGE_HEAP（）

closed 2017-11-01 by gylalahuo