ARMmbed/uvisor #526: the config of uvisor_box_heapsize can't set to 17*1024

gylalahuo, 2017-11-01:
when i set uvisor_box_heapsize(16*1024),no error. when i set uvisor_box_heapsize(17*1024),
 and compile the program ,the error log is:
c:/program files (x86)/gnu tools arm embedded/4.9 2015q1/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: address 0x20002080 of ./BUILD/K64F/GCC_ARM/mbed-os-example-uvisor-master.elf section `.uvisor.bss' is not within region `m_data'

so can i set the box_heapsize to 17*1024?

alzix, 2017-11-05:
Probably not, otherwise there won't be an error reported by the linker.
Refer to the linker script: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld
Apparently you have request too big chunk of memory for your box, so the resulting image won't fit in to the memory.
I would advise double checking whether you really need that much heap memory for your box.

gylalahuo, 2017-11-06:
Hi alzix，thanks for your answer!
In my graduation project， i ‘m trying to  combine mbedTLS and mbed uvisor to protect https commmunication。some errors i have sloved，some not。
For example，in https communication there need malloc 34*1024 memory in mbedtls_ssl_setup function.
because it's use malloc()  ,whatever heap size i set to secure_box,it always fails.then I find use secure_malloc(secure,16*1024) ，can success malloc 16K memory,but use 
	if(mbedtls_calloc(1,8*1024)==NULL||mbedtls_calloc(1,8*1024)==NULL),i failed. the mbedtls_calloc  use calloc function .so i try to replace mbedtls_calloc to secure_malloc() in mbetls_library. unfortunately，it halt in uvisor_api().  and i follow the uvsior debug ，can‘t see the reason why it halt。
Thanks ！

gylalahuo, 2017-11-06:
hi alzix 
the project is upload to github, the url is https://github.com/gylalahuo/test1.git.
now i only want realise https communicate with my server in secure box. the main problem is memory malloc.  in https communication there are many function use calloc() function.  it seems thart  use calloc() function  directly,can't get the memory i want.

mikisch81, 2017-11-06:
1. Actually you don't need to replace the mbedtls `calloc()` calls because if you enable uvisor then all the memory operations are swapped by the `secure_*()` calls, so calling `calloc()` will actually call `secure_calloc()`.
1. Wouldn't it be easier just to change the linker script of the target you are working on?
The linker error you encounter is because the `.uvisor_bss` section got to big because of the huge heap you gave for the https box which caused this section to overflow the boundries of the `m_data` memory area.
So you can increase the length of the `m_data` memory area and of course move the starting address of the `m_data_2` memory area to accommodate your memory requirements.
For example: the memory areas of K64F board: [https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld#L78:L86](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld#L78:L86)

gylalahuo, 2017-11-07:
Hi mikisch81 ，
i will try it on your advise。but i still have a probelm.  If i  call https function in my main.cpp ,there no error ，when i use  mbedtls_calloc() function to get 32K heap size. but if i call https function in  secure box in https.cpp , it can't get the corresponding heap size。
Thanks！

gylalahuo, 2017-11-08:
i have a question, what is the relation of 
UVISOR_SET_PAGE_HEAP(), UVISOR_BOX_HEAPSIZE(), secure_allocator_create_with_page() .
from the test.  UVISOR_BOX_HEAPSIZE() allocates memory seems to don't have relation of secure_create_with_page() , the memory allocated by secure_create_with_page() can bigger than UVISOR_BOX_HEAPSIZE().

gylalahuo, 2017-11-08:
hi  alzix ,mikisch81,
Thank you for your help.
 i have solved the memory allocate problem after do a lot of  experiments. but  there's another problem .
now i'm analyzing the reason for this problem .

closed 2017-11-08 by gylalahuo