ARMmbed/uvisor #531: use secure_malloc to allocate memory,this memory is secure?

gylalahuo, 2017-11-20:
for a secure_box, which memory is secure? I think the stack space belong to secure_box is secure,how about the heap allocated for secure_box?

Patater, 2017-11-21:
If you use `malloc` from a secure box (i.e. any box except the public box), the memory allocated will be private to the secure box that allocated it. uVisor does this by wrapping `malloc` to use the secure allocator internally.

closed 2017-11-21 by Patater

gylalahuo, 2017-11-22:
Hi Patater,
thanks for your answer.I still have a problem.
if the SecureAllocator   is  gloable object,like this

UVISOR_BOX_NAMESPACE("client_a");
UVISOR_BOX_HEAPSIZE(16*1024);
UVISOR_BOX_MAIN(my_box_main, osPriorityNormal, 8*1024);
UVISOR_BOX_CONFIG(my_box, acl, 1024*4, my_box_context);
SecureAllocator secure=secure_allocator_create_with_pages(54*1024,17*1024);

then the memory allocated by secure_malloc belongs to secure_box's private memory?

Patater, 2017-11-22:
No, in the global case, `secure_allocator_create_with_pages` runs in the context of the public box, which would mean the memory is available to all boxes.

Use `uvisor_ctx` to have "globals" that are private within a box.

gylalahuo, 2017-11-22:
so what i though is totally wrong。
the secure_malloc uses public _page_heap_reserved memory,and UViSOR_SET_PAGE_SIZE  function sets  public _page_heap_reserved. the public_page_heap is not secure.
~o~
so what should i do to increase secure box heap size. it seems UVISOR_BOX_HEAPSIZE function  has a maximum heap size limit  ,can't bigger than 17K,because when I set UVISOR_BOX_HEAPSIZE(17*1024),it will show that 
[image]

Patater, 2017-11-22:
Using `UVISOR_BOX_HEAPSIZE` to configure your private box is the right way to do it. Set the page heap size to as small as you can  (via `UVISOR_SET_PAGE_HEAP`), potentially down to zero, and increase the box heap size. Use ordinary `malloc` from the private box to do private allocations.

You can also consider using an Mbed OS configuration mechanism to reduce the size of the public (legacy) heap, to free up more space for other things.

gylalahuo, 2017-11-23:
Hi   Patater,
i did some experiments as you suggested.  
First i setted page heap size at 1K by UVISOR_SET_PAGE_HEAP(1*1024,1),and find that BOX HEAPSIZE can set at 20K(can't bigger),but it still fail to meet my request,i need 54K memory. Is there another way to solve this problem, or  is it impossible to set at 54K? 
in .ld file,
there two memory 
like 
{
m_data  (RW) : ORIGIN=0X1FFF000,LENGTH=0X00010000
m_data2 （RW）:ORIGIN=0X20000000，LENGTH=0X00030000
}
I think public_box_heap memory  gets from m_data2 regin, private box heap size gets from m_data regin? is it right.

Patater, 2017-11-23:
Yes, private box heap is going into `m_data` (via `.keep.uvisor.bss.boxes`). `m_data` is at most 64 KiB. It seems unlikely we can get a large enough box heap from that.

Your original page heap idea should work with a minor adjustment. Instead of making a global SecureAllocator, use the `uvisor_ctx` to store a pointer to the allocator and create the allocator in the private box main.

I haven't tested the below code, but something like the following should help explain.

```C
struct box_context {
    SecureAllocator secure;
};

static void private_ box_main(const void *);

UVISOR_BOX_NAMESPACE("private_box");
UVISOR_BOX_HEAPSIZE(3072);
UVISOR_BOX_MAIN(private_ box_main, osPriorityNormal, 1024);
UVISOR_BOX_CONFIG(box_private, acl, 512, box_context);

/* We want 64 KiB pages, to allow allocations of 54 KiB.
 * We don't need any more than a single page (single allocation).
 * Note that this is a global configuration, not per-box. */
UVISOR_SET_PAGE_HEAP(64 * 1024, 1);

void private_ box_main(const void *) {
...
    static const size_t total_size = 54 * 1024 + 54 * 100; /* Reserve 54 KiB plus some overhead */
    static const size_t maximum_malloc_size = 54 * 1024; /* You want 54 KiB at once, in a single allocation. */
    uvisor_ctx->secure = secure_allocator_create_with_pages(total_size, maximum_malloc_size);
...
}

gylalahuo, 2017-12-05:
Hi patater,
I used you method and the problem is solved.but i find  it doesn't secure the private box memory .
for example
the memory allocate by  secure_malloc(uvisor_ctx,1024) in private BOX can be accessed by public BOX. public box can read write the region which belong to private.