ARMmbed/uvisor #533: uvsior can‘t prevent private memory from reading or writing by public box

gylalahuo, 2017-12-05:
Does uvsior  not prevent private box memory from being aceessed by public BOX?
I do some test and find in public BOX ,I can write the memory which belongs to private BOX.
for example 
In my  private box ,there is a local variable name "a" like 
int a=2;
and the a located at 0X1FFF4254.
then In public box .i write 
int *b=(int *)0x1fff4254;
*b=20.
run the program I find the variable"a" is changed to 20. it seems that the public BOX can revise the memory regin which belongs to private BOX
so i think uvsior can‘t prevent private memory from reading or writing by public box.is it rigth?

gylalahuo, 2017-12-06:
How test the security when enable UVSIOR? I saw a ducument about UVISOR memory protection。
Like this：
[image]
In picture，different BOX can‘t’ access memory belongs to others BOX.  I think it means that 
 if in public BOX has a stack overflow ,it will not affect other memory regin.
but when i use memset(a,-1,56*1024),"a" belongs to public memory .and use gdb to see the memory .i find   the memory belongs to private box is also overwrited to -1.
I am not sure this test method is right. if i am right.what meaning about uvisor memory protection,because i seems that it can't provent memory overflow. if i am wrong ,would you please tell me how to test?
Thanks

orenc17A, 2017-12-10:
@gylalahuo, we couldn't reproduce the scenario you're referring to.
The issue you're experiencing could be caused by a large number of reasons
Few things to check:
-	Insure your ACL does not expose the whole RAM
-	Make sure uVisor is enabled via compilation macros